EtchedPixels/FUZIX #513: Z88 port?

spth, 2017-09-26:
Is anyone working on a port of FUZIX to the Cambridge Z88? IT seems to be a nice small portable Z80-based machine.

Philipp (who does not have a Z88, but might consider getting one if FUZIX ran on it)

beretta42, 2017-09-27:
Its a z80 cpu, the base model is 32k ram, so it probably needs ram
expantion to work with fuzix.

EtchedPixels, 2017-09-28:
The Z88 supports a lot of banked expansion and its existing OS was designed for the job and actually did remarkably well - it even grrew things like tcp/ip support. They are ridiculously pricy over here at least and considered pretty collectable.

The biggest Fuzix problem for Z88 would I think be storage. The Z88 was designed to use plug in memory mapped ROM and EPROM chips (it even supported early 12v erase flash). I've never seen anyone add a proper mass storage adapter to them. Even the 'retro era' big enhancement was a 1MB flash card with 5v erase that worked in all slots (so you could have 3MB of flash)

In theory nothing stops someone building an SD adapter but it simply wouldn't match up with the way OZ (the existing OS) handles storage and basically treats flash as a backup media.

Otherwise architecturally providing you've got some extra RAM or RAM cards (and you can trivially go to 128K internal, 512K with two soldered wires) it actually looks like a lot of other Z80 platforms 4 x 16K banks each bank with a register providing the upper bits for the 16K bank.

closed 2017-11-07 by EtchedPixels