EtchedPixels/FUZIX #514: userstructs.h path issue

spth, 2017-09-28:
I just cloned FUZIX, and did a "make PLATFORM=zx128". I get:

~~~~
CC .obj/zx128/util-cat/Applications/util/cat.rel
In file included from ./Library/include/unistd.h:6,
                 from Applications/util/cat.c:22:
./Library/include/syscalls.h:12:29: fatal error: sys/userstructs.h: No such file or directory
compilation terminated.
In file included from ./Library/include/unistd.h:6,
                 from Applications/util/cat.c:22:
./Library/include/syscalls.h:12:29: fatal error: sys/userstructs.h: No such file or directory
compilation terminated.
Applications/build.mk:36: die Regel für Ziel „.obj/zx128/util-cat/Applications/util/cat.rel“ scheiterte
make: *** [.obj/zx128/util-cat/Applications/util/cat.rel] Fehler 1
~~~~

Philipp

EtchedPixels, 2017-09-28:
David never finished or sorted out the revised build system so it does indeed break somewhat

You should be fine if you follow the older instructions and build the userspace with

cd Library
make
make install

then try and build Applications/utils etc

Fixing the build system more sanely is on my TODO list, but probably means making a tiny patch to SDCC to make it accept both ".rel" and the normal ".o" that every other compiler expects.

spth, 2017-09-28:
Up to "make install" this seems to work (when executing "make install" as root), but in Applications/utils I then get:

~~~~
philipp@notebook5:~/FUZIX/Applications/util$ make
../../Library/tools/fcc   banner.rel -o banner
?ASlink-Warning-Couldn't find library 'c'
?ASlink-Error-<cannot open> : "/opt/fcc/lib/crt0.rel"
Makefile:141: die Regel für Ziel „banner“ scheiterte
make: *** [banner] Fehler 1
~~~~

An indeed /opt/fcc/lib/ is empty.

Philipp

spth, 2017-09-29:
Also, build instructions seem a bit outdated.
https://github.com/EtchedPixels/FUZIX/blob/master/Kernel/README.md
mentions setting TARGET in a top-level Makefile, but it seems only the Makefile in Kernel/ uses it. It also talks about three different allocs-per-node lines, as if these were in the top-level or Makefile, but I only see those in Library/libs/Makefile and Kernel/cpu-z80/rules.mk

Philipp

EtchedPixels, 2017-09-29:
I took a bit of time to apply violence to the build logic and implement a simple top level Makefile process and the supporting bits in place of the clever stuff that never worked right.

You should now be able to set TARGET in the top Makefile and make clean; make

spth, 2017-09-29:
As root the build process fails since it can't find the command fcc (I tried the msx2 and z80pack targets). As non-root it fails earlier when trying to remove /opt/fcc/bin/binman

Philipp

P.S.: It would also be nice if building would not require root.

EtchedPixels, 2017-09-30:
It doesn't require root except to mkdir /opt/fcc; chown whoever /opt/fcc. That's not ideal but how I do it. I should indeed add  /opt/fcc/bin to the path in the Makeifle. Fixed that now.

For a build all the Z80 platforms build the same user space and most of the same kernel code so z80pack is probably the easiest and gives you an image you can also run on the emulator easily.

spth, 2017-09-30:
OK, I'll go with z80pack.

With fcc in the path, build now proceeds much further, and fails at

~~~~
make[1]: Leaving directory '/home/philipp/FUZIX/Library/libs'
(cd Applications; make)
make[1]: Entering directory '/home/philipp/FUZIX/Applications'
(cd util; make -f Makefile.z80)
make[2]: Entering directory '/home/philipp/FUZIX/Applications/util'
../../Library/tools/fcc   banner.rel -o banner
?ASlink-Warning-Couldn't find library 'c'
?ASlink-Error-<cannot open> : "/opt/fcc/lib/crt0.rel"
Makefile.z80:141: recipe for target 'banner' failed
make[2]: *** [banner] Error 1
make[2]: Leaving directory '/home/philipp/FUZIX/Applications/util'
Makefile:4: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/philipp/FUZIX/Applications'
Makefile:59: recipe for target 'apps' failed
make: *** [apps] Error 2
~~~~

Also, I assume the normal way to change the options passed to SDCC is to modify fcc?

Philipp

EtchedPixels, 2017-09-30:
Ok so closer but for some reason it missed out copying the crt0 and C library. Must be short an install somewhere.

The normal way to fiddle with the SDCC options is to tweak fcc. The goal of fcc is to make SDCC look a bit more like a traditional unix C compiler. I'll probably wrap cc65 and the GNU compilers with it in time to get all the other weirdness out of the Makefiles. They'll need less wrapping but still some.

Any particular sdcc options you had in mind ?

Try now...

spth, 2017-10-01:
Thanks. Build progresses much further now. After a while I apparently run into an SDCC bug:

~~~~
../../../Library/tools/fcc  -O2 -c test.c
test.c:105: warning 85: in function file_exists unreferenced function argument : 'stat'
test.c:206: error 129: pointer types incompatible 
from type 'void generic* fixed'
  to type 'void generic* fixed'
~~~~

I'd like to file a bug report for SDCC. Is there an easy way to make fcc dump the sdcc command line it invoces?

Philipp

P.S.: The SDCC option I wanted to try was --dump-graphs to collect some statistics that might help improve SDCC compilation time / code quality trade-off. It is probably not of interest to other FUZIX users.

EtchedPixels, 2017-10-01:
-V will dump the arguments being used

I've added the -Wc,-foo oddness that some other compilers have for passing stuff to given passes. It's not quite the way they use it but near enough and you can now do -Wc,--dump-graphs

spth, 2017-10-01:
SDCC bug filed at https://sourceforge.net/p/sdcc/bugs/2660/

Philipp

spth, 2017-10-02:
Isn't fcc esentially just another part of the FUZIX build infrastructure? I guess it would be nicer if building FUZIX would not require installation of fcc into /opt, and would instead just use an fcc from inside the FUZIX tree.

Philipp

EtchedPixels, 2017-10-02:
It's useful to have somewhere you can install it but eventually yes that needs sorting. More important things to do first though!

spth, 2017-10-05:
Building mostly works for me now.
However, on the first build (and for the first build after a "make clean") I see this error in teh build output:

~~~~
cc     tools/bin2z80.c   -o tools/bin2z80
cp hogs.txt hogs.txt.old
cp: cannot stat 'hogs.txt': No such file or directory
cpu-z80/image.mk:32: recipe for target 'fuzix.bin' failed
make[1]: [fuzix.bin] Error 1 (ignored)
tools/memhogs <fuzix.map |sort -nr >hogs.txt
head -5 hogs.txt
~~~~

Philipp

spth, 2017-10-05:
I noticed that a few .rel files remain after "make clean":

Applications/V7/cmd/ed.rel
Kernel/cpm-loader/cpmload.rel
Kernel/cpm-loader/fuzixload.rel

Philipp

EtchedPixels, 2017-10-13:
Fixed I hope

closed 2017-10-13 by EtchedPixels