EtchedPixels/FUZIX #516: Casting function to void *

spth, 2017-10-05:
In Applications/MWC/cmd/test.c, line 199ff, a void * is initialized by a function pointer.
void is an object type, not a function type. C only allows casts between object types and casts between function types.

A fix would be to replace the void * by a function pointer type, e.g. change:

~~~~
struct op {
	const char *op_name;
	void *op_func;
};
~~~~

to

~~~~
struct op {
	const char *op_name;
	void (*op_func)(void);
};
~~~~

Philipp

erkinalp, 2017-10-05:
All FUZIX targets have linear addresses. It works correctly in ones we use.

EtchedPixels, 2017-10-05:
@erkinalp This isn't true for 8086 if you have CS != DS, nor will it be true for 65C816 when B != X

@spth  I'll fix those when I get a moment, unless you want to send a patch to do it. Thinking about systems with separate I and D address space I now understand why C has that rule.. learn something new every day!

spth, 2017-10-05:
I'll try to fix it (also elsewhere in FUZIX should I notice any).

Philipp

spth, 2017-10-05:
I've fixed the issue at
https://github.com/spth/FUZIX

However, it seems, Applications/MWC/cmd/test.c is no longer included in a "make TARGET=z80pack" build (unlike a few days ago), so I didn't actually compile the fixed code.

Philipp

EtchedPixels, 2017-10-16:
Merged

closed 2017-10-16 by EtchedPixels