EtchedPixels/FUZIX #539: util/fsck: fails on rebuilding freelist

beretta42, 2017-10-28:
Something about rebuilding the freelist wonks out the filesystem where pass 3 is concerned... a fresh filesystem passing /standalone/fsck, finds many, many, many multiply linked blocks in pass3 of /util/fsck :(

EtchedPixels, 2017-10-28:
The two versions are basically the same code. I think you may need to have words with your gcc again 8)

beretta42, 2017-10-31:
Still debugging this.  it seems the inode get fudged up by pass 2.  Everythings orderly until i get to a wacky inode with obviously wrong block allocation...  block 0xbd30 is alloced twice (in same inode)..  'bd' is cpu's 'jsr'.

beretta42, 2017-11-06:
Tracted issue down to an overflow error in the drivewire driver... /dev/devdw.c and /platform/drivewire.s overflow when converting 512 byte blocks to 256 byte blocks.  A max sized fs will aslpode.

closed 2017-11-06 by beretta42