EtchedPixels/FUZIX #562: Issues trying to compile for Mark4 N8VEM

cocus, 2017-11-20:
Hello all. I'm trying to compile from the latest sources to the Mark4 N8VEM SBC, just to try the compilation process. The final goal is to take this as a template for my board, which was pulled out of a Tekelec data analyzer (telecom stuff).
I have a Linux Mint virtual box (18.2), which only contains the SDCC (compiled from sources, current ones as for this post). SDCC seems to be working fine, as I played around creating a small monitor for this HD64180 processor, and it works.

I made the appropiated edits to the Makefile to reflect the board I wanted to compile to (the mark 4), and after that, I ran make.
It took a lot of time, but it got stuck in stty.c; claiming that there was an error in the recipe for that target. It's in spanish (I forgot to change it to english), but it didn't gave more information. 
```
../../Library/tools/fcc  -O2 -c stty.c
stty.c:155: warning 196: pointer target lost const qualifier
Makefile.z80:137: fallo en las instrucciones para el objetivo 'stty.rel'
make[2]: *** [stty.rel] Error 1
make[2]: se sale del directorio '/home/cocus/z80/fuzix/FUZIX/Applications/util'
Makefile:10: fallo en las instrucciones para el objetivo 'util'
make[1]: *** [util] Error 2
make[1]: se sale del directorio '/home/cocus/z80/fuzix/FUZIX/Applications'
Makefile:60: fallo en las instrucciones para el objetivo 'apps'
make: *** [apps] Error 2
```
What can I to get a more verbose output?

I tried make -i which is not a good thing to do, but some errors appeared as well (exactly like those). And it couldn't build the .bin image at all.

Maybe I'm doing something wrong, or maybe I need another step prior? 

Thanks, and I hope this is the right place to ask.

EtchedPixels, 2017-11-21:
Makefile.z80:137: fallo en las instrucciones para el objetivo 'stty.rel'

Can you tell me what the exact makefile equivalent message is in English as I can't duplicate it at the moment and I don't see anything in there that would explain why it decided something had failed.

The other thing you can do is to add -V to the fcc command liine, then it will display the sdcc commands it runs.

EtchedPixels, 2017-11-21:
Can you tell me what that exact make message is in English. Also it might be worth checking it's not something silly like out of disk space or memory killing sdcc (sdcc will use obscene amounts of resources on some things!)

cocus, 2017-11-21:
Hi, sorry about the spanish thing. 
Just to clarify it says "recipe for target 'stty.rel' failed".

I increased the RAM size of the VM, and I can see SDCC taking up almost 4.5GiB of RAM. I think that was the issue, since it compiled stty.c without a hitch.

Now I'm having a similar issue with dc.c, but this time the linker is complaining the .rel file is not found:
```
make[2]: Entering directory '/home/cocus/z80/fuzix/FUZIX/Applications/V7/cmd'
../../../Library/tools/fcc  -O2 -c dc.c
dc.c:716: warning 24: index 100 is outside of the array bounds (array size is 100)
dc.c:731: warning 24: index 100 is outside of the array bounds (array size is 100)
../../../Library/tools/fcc   dc.rel -o dc
?ASlink-Error-<cannot open> : "dc.rel"
Makefile.z80:57: recipe for target 'dc' failed
make[2]: *** [dc] Error 1
make[2]: Leaving directory '/home/cocus/z80/fuzix/FUZIX/Applications/V7/cmd'
Makefile:13: recipe for target 'cmd' failed
make[1]: *** [cmd] Error 2
make[1]: Leaving directory '/home/cocus/z80/fuzix/FUZIX/Applications'
Makefile:60: recipe for target 'apps' failed
make: *** [apps] Error 2
```

I've added the -V to fcc (just for the V7/cmd/Makefile.z80) and here is the result:

```
../../../Library/tools/fcc -V   -O2 -c dc.c
[sdcc] [-V] [--std-c99] [-mz80] [--less-pedantic] [--max-allocs-per-node] [100000] [--opt-code-size] [-D__FUZIX__] [-Ddouble=float] [-I/opt/fcc/include/] [-c] [dc.c] 

+ /usr/local/bin/sdcpp -nostdinc -Wall -std=c99 -D__FUZIX__ -Ddouble=float -I/opt/fcc/include/ -obj-ext=.rel -D__SDCC_STACK_AUTO -D__SDCC_CHAR_UNSIGNED -D__SDCC_INT_LONG_REENT -D__SDCC_FLOAT_REENT -D__SDCC=3_6_9 -D__SDCC_VERSION_MAJOR=3 -D__SDCC_VERSION_MINOR=6 -D__SDCC_VERSION_PATCH=9 -D__SDCC_REVISION=10182 -D__SDCC_z80 -D__STDC_NO_COMPLEX__=1 -D__STDC_NO_THREADS__=1 -D__STDC_NO_ATOMICS__=1 -D__STDC_NO_VLA__=1 -D__STDC_ISO_10646__=201409L -D__STDC_UTF_16__=1 -D__STDC_UTF_32__=1 -isystem /usr/local/bin/../share/sdcc/include/z80 -isystem /usr/local/share/sdcc/include/z80 -isystem /usr/local/bin/../share/sdcc/include -isystem /usr/local/share/sdcc/include  dc.c 
dc.c:716: warning 24: index 100 is outside of the array bounds (array size is 100)
dc.c:731: warning 24: index 100 is outside of the array bounds (array size is 100)
../../../Library/tools/fcc -V    dc.rel -o dc
[sdcc] [-V] [--std-c99] [-mz80] [--no-std-crt0] [--nostdlib] [--code-loc] [256] [--data-loc] [0x0] [--less-pedantic] [--opt-code-size] [-D__FUZIX__] [-Ddouble=float] [-I/opt/fcc/include/] [-L/opt/fcc/lib/] [-odc] [/opt/fcc/lib/crt0.rel] [dc.rel] [-lc] 

+ /usr/local/bin/sdldz80 -nf dc.lk
?ASlink-Error-<cannot open> : "dc.rel"
+ /usr/local/bin/sdldz80 -nf dc.lk returned errorcode 768
```

Oh and finally, I have about 70GB free in the VM.

Do I need more RAM? I currently have 8G in the host system, and using more than 4 in the VM will be kind of bad for the host.

Thanks!

EtchedPixels, 2017-11-22:
I'm not sure how much you actually need and how much can be swap. My box has 96GB so I've not really checked. If it does bomb out of memory then there ought to be a SIGKILL to it and I tihnk a log in dmesg.

If you take off the -O2 for problem apps it'll dramatically reduce memory and time usage . You'll see some makefiles already have things tthey don't try and optimize because they take forever.

I know Philipp was looking at the SDCC graphis and profiling/collecting data on the Fuzix build so hopefully it'll improve as he improves SDCC.

There's also a bug in fcc here - it doesn't check if the child died due to a signal and error itself. I'll fix that one.

EtchedPixels, 2017-11-22:
Pushed an fcc fix that will hopefully let you see what is going on

EtchedPixels, 2017-11-22:
If it helps I mostly use sdcc 3.6.0 #9610

cocus, 2017-11-22:
I'll take a look at your latest commit. Do I have to do anything to uninstall the current version? Like make uninstall (or remove)?
In the meantime, I'll try to remove the -O2 from the V7/utils

trothr, 2017-11-22:
Not specific to FUZIX, I've been pushing a message handler that equates "recipe for target 'stty.rel' failed" and "fallo en las instrucciones para el objetivo 'stty.rel'" in different national repositories. You just swap out the repo to suit local needs. (Not everyone is multi-lingual.) Full message codes let you correlate with your own tongue even if another language is displayed. 

Now where did I put that GitHub project? ...

EtchedPixels, 2017-11-22:
@cocus it should just build the newer fcc and install it if you update your git tree and make

cocus, 2017-11-23:
So, I'm currently trying this in another mint 18.2 VM (since this is the one that I have at the moment, but it has 2GB of ram, and 60G of hard drive free space). 
I've used the #9610 revision of sdcc and #7e1839d of fuzix. This time it went better until the .bin image creation. 
```
tools/bankld/sdldz80 -n -k /usr/local/share/sdcc/lib/z180 -f platform-n8vem-mark4/fuzix.lnk
cp hogs.txt hogs.txt.old
cp: cannot stat 'hogs.txt': No such file or directory
cpu-z180/image.mk:26: recipe for target 'fuzix.bin' failed
make[1]: [fuzix.bin] Error 1 (ignored)
tools/memhogs <fuzix.map |sort -nr >hogs.txt
```
Regardless, the .bin was created. And it didn't stop on dc.c as before. Maybe the key was the revision of the SDCC. I'll have to try it in my own VM.

I had some errors when compiling, but I ran make several times and each time it did got further and further. Those errors where triggered by some applications. 

Finally, I had some errors when building netd. The file sys/drivewire.h wasn't found and thus netd compilation ended right there; which actually stopped the build started via make all.

```
make[2]: Entering directory '/home/santiago/z80/FUZIX/Applications/netd'
fcc  -DNETD_LITTLE_ENDIAN -c netd.c
netd.c:37:27: fatal error: sys/drivewire.h: No such file or directory
compilation terminated.
Makefile.z80:22: recipe for target 'netd.rel' failed
make[2]: *** [netd.rel] Error 1
make[2]: Leaving directory '/home/santiago/z80/FUZIX/Applications/netd'
Makefile:37: recipe for target 'netd' failed
make[1]: *** [netd] Error 2
make[1]: *** Waiting for unfinished jobs....

```
Anyway, and a little bit offtopic, is there any IRC channel available to chat about fuzix? I might need some advice when porting this to the board that I have. I saw another issue (#564) that aims to create a proper porting manual, but in the meantime this would help.

And finally, sorry about the spanish text in my first post.

EtchedPixels, 2017-11-23:
I'm sometimes on zeniv.linux.org.uk 6667 as _A_Z80_  (and that's a Fuzix box running my IRC client!)

The google groups list is probably the best place to ask and I can send you an invite. Some of us also hang out on retrobrewcomputers.org whose forum might interest you if doing your own Z80 board.

What doe your platform look like. The memory paging tends to define how to do a port

cocus, 2017-11-23:
That would be great @EtchedPixels, I would like to join that list. I couldn't find it, so an invite might be necessary. 

And regarding this issue, I've tried with SDCC 9610 in my VM and it went away. It compiles the same way as in the other VM that i've tried earlier. So I guess this should be related to newer sdcc versions.
We should close this (the netd error might be a thing for another issue, not this one).
Thanks!

EtchedPixels, 2017-11-23:
Can you send an email to etchedpixels @ gmail.com so I know your address

EtchedPixels, 2017-11-23:
I'll try a new sdcc at some point and see what blows up - I normally test when Philipp announces the first release candidates for a release and then file bugs based on what goes boom.

closed 2017-11-24 by cocus