EtchedPixels/FUZIX #568: reboot leaves filesystem dirty

tormodvolden, 2017-11-25:
Running reboot (also without -f) leaves the filesystem dirty and it will be mounted read-only on next boot. I can run "remount $ROOTDEVICE ro" before reboot, and it will be fine. Must this be handled by init scripts, or should the kernel unmount it cleanly?

BTW, after running remount, prtroot still report rw.

EtchedPixels, 2017-11-25:
On Sat, 25 Nov 2017 18:38:18 +0000 (UTC)
Tormod Volden <notifications@github.com> wrote:

> Running reboot (also without -f) leaves the filesystem dirty and it will be mounted read-only on next boot. I can run "remount $ROOTDEVICE ro" before reboot, and it will be fine. Must this be [...]

init scripts

In your boot up boot with ro option
make /etc/rc run fsck / then fsck and mount any other fs (need to do
fsck -a)

My init 6 at the moment contains nothing but a script to run which does
umount -a remount / ro

Alan

beretta42, 2017-11-25:
Fwiw:

'reboot' just calls sync and calls the platform's reboot implementation.
It alone doesnt safely shut down procs/disks.  I usually:

# shutdown
telinit 1
sleep 5
# umount -a
remount / ro
echo shutdown.
shutdown

Usually gets me close, with an occassional "dev busy" on remount (probably
some open file somewheres)

i do think we need what Alan called for, a 'killall' that will slaughter
things in odd pd's.  I notice linux's 'shutdown' binary seems to be the
above script added with a sys5 'killall', all-in-one.

closed 2017-12-04 by EtchedPixels