EtchedPixels/FUZIX #574: Excessive RAM support

HenryDane, 2017-12-17:
Is there a way to configure FUZIX's memory system to allow for banking of entire 64K segments? I am working on a system with 288K SRAM, and 32K EEPROM. Is this already a feature?

beretta42, 2017-12-17:
Check out the bank16 memory model. This can get me around 60k user spaces.
Probably could do even more, but for a Color Computer, we need the
interrupt handler to be around.

brett

EtchedPixels, 2017-12-17:
You need either some common space or a way to write things into a bank you are not executing. So for example

6509 is a 6502 with 64K bank switching and the ability to write into another bank, and that is sufficient· 65C816 has bank registers that mostly work on 64K boundaries but allows you to do far calls and reads/writes, and that is supported today.
The Cromenco Z80 systems had 64K banks but you could also set it to write to all banks at once so at boot up an OS would do something like

set writes to all banks
ld hl, common
ld de, common
ld bc, common_size
ldir
set writes to one bank

And I'm fiddling around with 68HC11 support on a design where you have 256 bytes of common memory (the on chip RAM) which is sufficient to set up a common space and bank interrupts.

Basically you need a way of getting a common stub of code into each user process bank. If you can do that then you are sorted.

HenryDane, 2017-12-17:
Thanks for the help. I will try that out.

closed 2017-12-17 by HenryDane