EtchedPixels/FUZIX #593: Remounting root ro fails

beretta42, 2018-03-01:
the kernel's umount call checks for writers and on-disk link count for 0, otherwise it fails:

https://github.com/EtchedPixels/FUZIX/blob/master/Kernel/syscall_other.c#L398

In my particular case, "/etc/mtab" is the culprit inode that's bombing out umount(). The kernel is reporting no writers on the file, and is returning a nlink of 1.  Should the 'remount' utility be unlinking /etc/mtab before trying to umount() it?  Or maybe umount() should have the OR logic flipped to AND logic or something?

beretta42, 2018-03-02:
Hmm... maybe we're catching the defunct inode.  Should a check for the inode actually being valid be done as well?

beretta42, 2018-03-04:
Nah.  That code is right.  Something is holding c_refs open on /etc/mtab.  I suppose it is during 'remount' is when this is all happening.  maybe rename() needs checked.

beretta42, 2018-03-04:
yup.  rename() was holding on to dsti.  This seems to work: 

`https://github.com/beretta42/FUZIX/commit/12ba565a34c94b321fabb89070f486a1321142be`

Although I don't trust my understanding here enough to do a PR and know I wouldn't be screwing things up worse :)

EtchedPixels, 2018-03-09:
Fixed I think ?

closed 2018-03-09 by EtchedPixels

beretta42, 2018-03-09:
Yes.   seems to be in order