EtchedPixels/FUZIX #594: Debug 6809 floating point  support

EtchedPixels, 2018-03-09:
We build almost all the relevant library code now but there are a few that don't built and we also don't seem to get correct answers for many cases.

beretta42, 2018-03-10:
[image]

Disregard the quick and dirty float printf;  its a start.

EtchedPixels, 2018-03-11:
Looks like a good start for floats. Double will be the one that's going to be buggy I suspect - and because of the way ANSI C is defined a float passed in varargs gets turned into a double.

beretta42, 2018-03-15:
We got gcc NOT to issue any 64 bit calls., successfully building code with "double". (Which is still 32 bits, btw) Many math.h functions work, but powf() and pow() go into la-la land and crash fuzix.  We either broke something or found a sleeper bug.

EtchedPixels, 2019-06-02:
Seems to work better for me now I've taught it that double is 32bit not 64bit (oops)

beretta42, 2019-06-13:
I was working on cleaning up compiler warnings and found many warnings regarding shifting a 32 int 63 times.

EtchedPixels, 2024-06-19:
Deferring future work on this until the new compiler