EtchedPixels/FUZIX #598: bug: ucp not making /bin

beretta42, 2018-04-02:
Some flakiness is ensuing while building the filesystem.  My best guess so far is the caching isn't quit working.

```
Standalone $ ./mkfs -X filesystem-src/fuzixfs.dsk 256 65535
Making 512 byte/block filesystem with reversed byte order on device filesystem-src/fuzixfs.dsk with fsize = 65535 and isize = 256.
Opening filesystem-src/fuzixfs.dsk (offset 0)
Standalone $ ./ucp filesystem-src/fuzixfs.dsk 
Opening filesystem-src/fuzixfs.dsk (offset 0)
Fuzix UCP version 1.3ac. Type ? for help.
unix: mkdir bin
unix: chmod 0755 bin
chmod bin to 0755
chmod: can't stat file 'bin': 2
unix: ls
drwxrwxrwx   2 13056         512 Apr 02  118   14:01  ./             
drwxrwxrwx   4   256         512 Apr 02  118   14:01  ../
```

EtchedPixels, 2018-04-02:
Bug is in mkfs
(try mkfs/fsck/ucp)

beretta42, 2018-04-03:
ok that combination works.  A pull of yesterday's mkfs clarification seems to make it work again - but I don't see why.

EtchedPixels, 2018-04-04:
Unless I'm missing something it's a compiler bug in gcc

closed 2018-05-13 by EtchedPixels