EtchedPixels/FUZIX #616: z80pack: "y.tab.c:1498: error 101: too many parameters" FTBFS

piso77, 2018-05-18:
$ sdcc -v
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.6.0 #9615 (Linux) published under GNU General Public License (GPL)

$ git log --oneline -1
c938656 trs80: switch over to using the bank94 code

$ make
...
make[2]: Leaving directory '/home/flag/FUZIX/Applications/netd'
(cd SmallC; make -f Makefile.z80)
make[2]: Entering directory '/home/flag/FUZIX/Applications/SmallC'
Makefile.z80:52: warning: overriding recipe for target 'clean'
Makefile.z80:44: warning: ignoring old recipe for target 'clean'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/flag/FUZIX/Applications/SmallC'
(cd MWC/cmd; make -f Makefile.z80)
make[2]: Entering directory '/home/flag/FUZIX/Applications/MWC/cmd'
../../../Library/tools/fcc -mz80  -O2 -c find.c
y.tab.c:1233: warning 112: function 'yylex' implicit declaration
y.tab.c:1498: warning 112: function 'yyerror' implicit declaration
y.tab.c:1642: warning 112: function 'yyerror' implicit declaration
y.tab.c:1498: error 101: too many parameters 
y.tab.c:1642: error 101: too many parameters 
find.y:216: warning 85: in function yyerror unreferenced function argument : 'p'
-:0: warning 85: in function fentry unreferenced local variable : 'nb'
-:0: warning 85: in function fentry unreferenced local variable : 'fd'
-:0: warning 85: in function fentry unreferenced local variable : 'iobuf'
find.y:520: warning 85: in function ffork unreferenced function argument : 'sbp'
-:0: warning 85: in function xprint unreferenced function argument : 'np'
-:0: warning 85: in function xnop unreferenced function argument : 'np'
Makefile.z80:43: recipe for target 'find.rel' failed
make[2]: *** [find.rel] Error 1
make[2]: Leaving directory '/home/flag/FUZIX/Applications/MWC/cmd'
Makefile:46: recipe for target 'MWC' failed
make[1]: *** [MWC] Error 2
make[1]: Leaving directory '/home/flag/FUZIX/Applications'
Makefile:61: recipe for target 'apps' failed
make: *** [apps] Error 2

piso77, 2018-05-20:
$ yacc --version
bison (GNU Bison) 3.0.4
Written by Robert Corbett and Richard Stallman.

Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Uhm, i guess that is my problem.

beretta42, 2018-05-20:
Works for my yacc, which is the same bison version as piso77's.

piso77, 2018-05-30:
Yep, and even moving to bsdyacc didn't fix it: and this is not the only build problem that i have...

Exactly, which system / chroot shall i use to build the z80pack target?

My host system is an amd64 ubuntu 16.04  with sdcc 3.6.0.

EtchedPixels, 2018-05-30:
I'm building on Fedora 27 and on Fedora 28
Currently using SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/TININative/ds400/hc08/s08/stm8 3.7.0 #10228 (Linux)
byacc-1.9.20170709-4.fc28.x86_64

Note the sdcc if you use 3.7.x you need at least 10228 or so because the base 3.7.0 doesn't build some bits correctly and the fix didn't quite make 3.7.0.

piso77, 2018-05-31:
Ok, i updated my sdcc:

flag@harukaze:~$ sdcc -v
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.7.1 #10414 (Linux)
published under GNU General Public License (GPL)

and now the compilation breaks in another point:

(cd V7/cmd; make -f Makefile.z80)
make[2]: Entering directory '/home/flag/FUZIX/Applications/V7/cmd'
../../../Library/tools/fcc -mz80  -O2 -c dc.c
dc.c:716: warning 24: index 100 is outside of the array bounds (array size is 100)
dc.c:731: warning 24: index 100 is outside of the array bounds (array size is 100)
used unbound variable in replacement
dc.c:856: warning 84: 'auto' variable 'dig' may be used before initialization
dc.c:919: warning 60: function return value mismatch
from type 'void generic* literal'
  to type 'int fixed'
dc.c:1118: warning 24: index 100 is outside of the array bounds (array size is 100)
dc.c:1123: warning 24: index 256 is outside of the array bounds (array size is 256)
dc.c:1143: warning 85: in function onintr unreferenced function argument : 'sig'
dc.c:1387: warning 84: 'auto' variable 'r' may be used before initialization
dc.c:1387: warning 84: 'auto' variable 'r' may be used before initialization
dc.c:1387: warning 84: 'auto' variable 'r' may be used before initialization
dc.c:1388: warning 84: 'auto' variable 'r' may be used before initialization
dc.c:1591: warning 85: in function hexot unreferenced function argument : 'flg'
dc.c:2073: warning 85: in function garbage unreferenced function argument : 's'
Makefile.z80:47: recipe for target 'dc.rel' failed
make[2]: *** [dc.rel] Error 1
make[2]: Leaving directory '/home/flag/FUZIX/Applications/V7/cmd'
Makefile:13: recipe for target 'cmd' failed
make[1]: *** [cmd] Error 2
make[1]: Leaving directory '/home/flag/FUZIX/Applications'
Makefile:61: recipe for target 'apps' failed
make: *** [apps] Error 2

I'll see if i can figure out what is going on.

EtchedPixels, 2018-05-31:
Thanks - sdcc has grown another correctness thing to fuss about - and it's completely correct about it.

Patch queued.

piso77, 2018-06-01:
The weird thing is that it doesn't spit out any error - do you passo any special flag to the toolchain or what? Because i would like to try fix some of these, but i don't see any error message...

Anyway, later on it break again (and again no error is showed):

...
make[2]: Entering directory '/home/flag/FUZIX/Applications/netd'
fcc -mz80  -DNETD_LITTLE_ENDIAN -c uip.c
uip.c:1057: warning 24: index 8 is outside of the array bounds (array size is 8)
uip.c:1218: warning 24: index 10 is outside of the array bounds (array size is 10)
Makefile.z80:22: recipe for target 'uip.rel' failed
make[2]: *** [uip.rel] Error 1
make[2]: Leaving directory '/home/flag/FUZIX/Applications/netd'
Makefile:40: recipe for target 'netd' failed
make[1]: *** [netd] Error 2
make[1]: Leaving directory '/home/flag/FUZIX/Applications'
Makefile:61: recipe for target 'apps' failed
make: *** [apps] Error 2

and this time around, before erroring out, it eats >32GB of memory...

EtchedPixels, 2018-06-01:
Do you have an out of memory error in "dmesg" ?

That would be a bug in whatever compiler of the day you grabbed 8( - file an SDCC bug and try a different snapshot. Turning down optimisation levels may help too (fcc probably does set them excessively high for the current compiler as it seems to do far better than the old one where it was the only way to get vaguely decent code)

SDCC is quite memory hungry and as I have 96GB of RAM (cheap old Xeon server 8)) I've not gotten around to tuning fcc. I'll add a bug for that so I remember to take a good look at it with the current sdcc

EtchedPixels, 2018-06-01:
Duplicated with todays sdcc - I'll file an sdcc bug... you need 32.5GB of RAM to build it even unoptimized which is a bit crap !

piso77, 2018-06-05:
Ok, i hit an ICE today:

$ sdcc -v
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.7.1 #10419 (Linux) published under GNU General Public License (GPL)

...
../../../Library/tools/fcc -mz80   -c calendar.c
calendar.c:233: warning 84: 'auto' variable 'retval' may be used before initialization
calendar.c:321: warning 84: 'auto' variable 'nfiles' may be used before initialization
calendar.c:357: warning 84: 'auto' variable 'foundfiles' may be used before initialization
calendar.c:388: warning 84: 'auto' variable 'matchstr' may be used before initialization
calendar.c:391: warning 84: 'auto' variable 'matchstr' may be used before initialization
calendar.c:403: warning 84: 'auto' variable 'matchstr' may be used before initialization
calendar.c:406: warning 84: 'auto' variable 'matchstr' may be used before initialization
calendar.c:416: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:416: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:416: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:416: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:431: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:432: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:437: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:443: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:448: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:453: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:457: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:462: warning 84: 'auto' variable 'matchdate' may be used before initialization
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '3632' : Shouldn't push BC if it's wiped out by the return 
Contact Author with source code
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '4336' : code generator internal error 
Contact Author with source code
[... cut ...]
calendar.c:469: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '4509' : code generator internal error 
Contact Author with source code
Makefile.z80:32: recipe for target 'calendar.rel' failed
make[2]: *** [calendar.rel] Error 1
make[2]: Leaving directory '/home/flag/FUZIX/Applications/MWC/cmd'
Makefile:46: recipe for target 'MWC' failed
make[1]: *** [MWC] Error 2
make[1]: Leaving directory '/home/flag/FUZIX/Applications'
Makefile:61: recipe for target 'apps' failed
make: *** [apps] Error 2

i filled a sdcc bug for this: https://sourceforge.net/p/sdcc/bugs/2756/

piso77, 2018-06-11:
Updated today's sdcc (after they fixed that memory hogs error), just to find another one:

https://sourceforge.net/p/sdcc/bugs/2759/

EtchedPixels, 2018-06-11:
Yep - normally Fuzix doesn't need a non release SDCC - when it does it gets a bit too exciting sometimes

piso77, 2018-06-14:
And another one:

https://sourceforge.net/p/sdcc/bugs/2761/

i was happily compiling FUZIX for the first time *ever* with 10438, when i saw a couple of small fixes in the commit logs and i decided to try update sdcc to 10444.

piso77, 2018-06-14:
Ok, i bisected it down to #10442 - if you want to rebase your sdcc branch, use #10441, that's the last one that completed a FUZIX build.

piso77, 2018-06-27:
3.7.1-trunk-r10445 have been working for me for the last few weeks, so i'm closing this.

EtchedPixels, 2018-06-27:
I need to rebase my sdcc + patches to 10445 or higher.