EtchedPixels/FUZIX #649: Relocatable binaries

EtchedPixels, 2018-07-21:
Various things need relocatable binaries so it makes sense to add support for them to the toolchains that don't yet handle it. cc65 does (but the format is sucky so wants a convertor), sdcc doesn't (but it's easy to add), gcc m6809 - who knows 8), 68000 already supported, 8086 - has segment bases anyway.

In addition we should change the binary format header so that everything is in 256 byte pages not in bytes (makes all the overflow maths trivial), and add better planned flags for compatibility and CPU type - for example 6309 v 6809, 6502 v 65C02 v 65C816.

Finally some flags for positioning hits because some platforms would work far better if the binary loader knew this binary wanted to use graphics and a bit of info about them.

All this makes possible dynamically binding to 'shared' libraries (which except for 6502 are probably not shared at all but save disk), and more importantly being able to bind to a graphics library for the machine at runtime.

EtchedPixels, 2018-10-04:
Relocation enabled for Z80/Z180 systems. There is an open question about whether enough of them have relocations that expand the size to justify doing a brk() back to the right boundary in crt0.s

No library relocation support yet. Thus the CP/M emulator is not yet able to relocate the high code

feilipu, 2018-10-05:
The issue of relocatable binaries (at least for z80 / z180) is interesting, for me in the [context of MP/M](https://github.com/z88dk/z88dk/issues/935#issuecomment-418688227).

It looks like the DRI PRL file type is a sufficient solution for at least z80 page relocation, and it is supported by the DRI LINK-80.

@EtchedPixels do you have an opinion as to whether the `.PRL` is a sufficiently good solution for building relocatable applications for Fuzix, and / or MP/M?

EtchedPixels, 2021-07-17:
Done 6502

EtchedPixels, 2021-08-18:
Status update:
6303/6803 is done but may have lurking bugs
6502 is done
6800 will be too slow
68HC11 is done
68000 is done
8080/8085 it's just too slow and all the platforms use 0x0100 anyway
8086 doesn't need it
ARM is done
EPS8266 doesn't need it (all ESP8266 are the same)
Z80/180/eZ80 is done

6809 still needs fixing

EtchedPixels, 2022-02-05:
And done for 6809.

closed 2022-02-05 by EtchedPixels