EtchedPixels/FUZIX #654: Unclear status of Thomson machines at https://github.com/EtchedPixels/FUZIX/wiki

pulkomandy, 2018-08-19:
Ok, I should probably sit down and actually get things to work instead of complaining but, ...

https://github.com/EtchedPixels/FUZIX/wiki mentions "Thompson DCM". I assume this was meant to be Thomson (no p), and refers to the MO6 and TO8 machines. "DC" are the initial of the person writing one of the emulators for these, and I think has nothing to do here.

Also, these machines have 2x16K of bankable memory windows, which can't be said to be small, right?

The main problem would be that the machines have ROM at FFxx, and that the interrupt handler jumps into the ROM.

EtchedPixels, 2018-08-20:
I clobbered most of that page and I've reworked the Thomson entry in the platform review page plus pointed stuff there to avoid duplication.

Given the map you have it sounds like you can run with

Kernel mode

0000-3FFF Kernel 
4000-5FFF Video | Kernel
6000-9FFF Not used
A000-DFFF Kernel 

User 
0000-3FFF Kernel |  Bank to copy data into 6000-9FFF for user apps
4000-5FFF Video | Kernel
6000-9FFF Copy from banked memory of current running process low space
A000-DFFF User high bank

I guess it all depends what the ROM irq handlers expect/touch and do before you can get hold of them.

A fancier arrangement would be to keep some common code copied into various banks and do stack switching games. Then you'd be able to get to something like

0000-3FFF  User | Kernel depending on mode | Video
4000-9FFF User (copied on task switch)
A000-DFFF User | Kernel | Buffers

It would be very tight to get the kernel into 32K plus banking the buffers and tty buffers and maybe some other low level bits separately (eg the video handling) and need much scarier system call and interrupt handling.

pulkomandy, 2018-08-20:
I was looking for the platform review page and found both https://github.com/EtchedPixels/FUZIX/wiki/Platform-Notes and https://github.com/EtchedPixels/FUZIX/wiki/Platform-Review-Notes . These can be merged, I think?

The IRQ handler will only need its direct-page which is located at 6000-60FF, but I guess that's a problem for the second mapping.

tormodvolden, 2018-08-20:
A couple of years ago I looked at the TO8 memory map and thought about how it could be handled (I was looking into buying a TO8 at the time but hadn't have time to chase this any further). So here's a further suggestion.

If you tandem-bank the two 16 spaces, you could do it pretty much like on the dragon-nx32 platform (bankfixed.c memory model), which has a 32K space resident (for common and one kernel part) - which you could use your 24K space for, and a 32K banked space for either second kernel part or for userspace - which you could use your two 16KB spaces for. Especially as the video memory can be in the banked part, the 24K (+2 x 16K banked) is plenty for the kernel.

You would only need to build your applications with a hole in the middle (often inside a code section, unless you manually split each application...). Back then, I made a patch for lwlink to try this out. I used here a hole at A000-B000 for testing but you can simply change that to 4000-A000. Then make _execve read the program in two blocks. The fun part is run-time RAM allocation, but restricting the heap to the upper 16K block should work fine in many cases.

This would be an easy way to get started, although less hacky ways could be investigated (also bank16.c?). Alan's suggestion of copying into the non-banked space would avoid the above tricks but at the cost of the copy in/out time (HD6309 would help here).
[hardcoded-gap.patch.txt](https://github.com/EtchedPixels/FUZIX/files/2302756/hardcoded-gap.patch.txt)
(funny that github won't let me attach a .patch file...)

EtchedPixels, 2018-08-20:
On the ZX spectrum 128K when I did that as a test, and on the MSX1 test I found the copying isn't that bad. In normal single user use we just don't task switch that often (and remember the tiny ports task switch by swapping to hard disk).

There are some awkward cases (like select) but the kernel goes to great length to do perfect wakeups for those.

pulkomandy, 2018-08-20:
If I go with HD6309, I may as well replace the ROMs to get away with the interrupt issues and... why would I even bother using existing/aged hardware then? :)

pulkomandy, 2018-08-20:
Yes, the ROM has font bitmaps somewhere. Not sure if it's easily reachable at a fixed address (there may be multiple revisions of the ROM). The floppy controller ROM may be useful (it provides sector access routines, and the THMFC1 is not a very high level floppy controller so it's nice to have these readily available).

I think the ROM will not handle video anywhere else than 4000-5FFF, so it will not be possible to use the existing textmode routines if we point vram to banks.

EtchedPixels, 2018-08-20:
I'd bet if you look for a complex character like say '$' that the 8 byte sequence only occurs once in the ROM in one place !

Quite a few kernels map the video memory in only when using it. That bit's not usually too hard except for the way it interacts with interrupt handling. You just make sure the video code isn't under the video memory.

pulkomandy, 2018-08-20:
Yes, the problem is the RAM at 4000-5FFF is not bankswitchable, so if you use this area for video (with the ROM routines), you can't use it for code as well.

And if you move the video to a bank (then mapped at 0000-3FFF, for example), then you can't use the ROM routines for text drawing. But that's fine, we can have our own code for this. In that case, 4000-5FFF is available for kernel code or whatever we want to put there.

EtchedPixels, 2018-08-20:
If the other areas of kernel are both banked I would put the common memory routines and stacks there. That makes life a lot less painful when bank switching or fielding interrupts.

closed 2018-08-31 by EtchedPixels