EtchedPixels/FUZIX #660: Teach fsck about sockets

EtchedPixels, 2018-09-07:
We can get stale socket inodes and right now we clean them up fine but report them as not the correct type and ask the user to Zap them

protoCall7, 2019-10-04:
@EtchedPixels I came across this issue while looking for a simple bug to get started with FUZIX development on. I believe I have a fix for this issue, but I was wondering if you knew of a good way to generate a stale socket inode so that I can test my changes before submitting a pull request?

EtchedPixels, 2019-10-05:
I think the simplest way may be to use mknod as root to create fake ones

            mknod("/tmp/asocket", 0140700, 0);
            unlink("/tmp/asocket");
            uadmin(A_REBOOT, 0, 0);

or maybe even just hack ucp so you can add some ?

protoCall7, 2019-10-14:
Thanks for the input, @EtchedPixels, the snippet above seems to do the trick!

On a related note, when booting with an improperly unmounted disk, is there a way to get a single-user mode, or another method of doing something similar to `yes | fsck -a` so that I don't have to spend as much time hitting "y <enter>" in order to test my changes?

EtchedPixels, 2019-10-14:
You can change the /etc/rc scripts for one

closed 2024-09-26 by EtchedPixels