EtchedPixels/FUZIX #661: Trying to call functions where there is no declaration visible in y.tab.c

spth, 2018-09-16:
Since SDCC 3.8.0 RC1 will probably arrive next week, i tried to compile current FUZIX with current SDCC from trunk. However, compilation fails and it doesn't look like the problem is on the SDCC side:

~~~~
…
../../../Library/tools/fcc -mz80   -O2 -c find.c
y.tab.c:1233: warning 112: function 'yylex' implicit declaration
y.tab.c:1498: warning 112: function 'yyerror' implicit declaration
y.tab.c:1642: warning 112: function 'yyerror' implicit declaration
y.tab.c:1498: error 101: too many parameters 
y.tab.c:1642: error 101: too many parameters
…
~~~~

Philipp

EtchedPixels, 2018-09-16:
What yacc tool is your system using for the build ?

It all works fine for me (Fedora) where yacc resolves to the one true yacc (Berkeley) but I've no idea what happens if your yacc resolves to bison.

EtchedPixels, 2018-09-16:
I'll give RC1 a spin once it's announced

spth, 2018-09-16:
bison 3.0.4, the current default in Debian GNU/Linux testing.

Philipp

EtchedPixels, 2018-09-16:
Bison isn't suitable for the Fuzix cases - it's a bit weird as it's generating stuff to run on an 8bit micro and bison really isn't good for that - even if I fixed up the prototypes

I'll update the docs somewhere

piso77, 2018-09-16:
From .travis.yaml:

...
    - sudo apt-get install -y byacc
    - sudo update-alternatives --set yacc /usr/bin/byacc
...

that should fix you compilation problem.

BTW, that reminds me i wanted to setup a nightly build using the trunk of
sdcc / cc65 /etc to catch toolchain issues in advance...

EtchedPixels, 2018-09-16:
added to INSTALL

Thanks

closed 2018-09-16 by EtchedPixels