EtchedPixels/FUZIX #669: error on ubuntu 18.10

jayacotton, 2018-10-25:
jay@jay-Ultra-27:~/projects$ sudo add-apt-repository -y ppa:p-pisati/fuzix
Get:1 http://security.ubuntu.com/ubuntu cosmic-security InRelease [79.6 kB]
Ign:2 http://ppa.launchpad.net/p-pisati/fuzix/ubuntu cosmic InRelease          
Hit:3 http://us.archive.ubuntu.com/ubuntu cosmic InRelease                                          
Err:4 http://ppa.launchpad.net/p-pisati/fuzix/ubuntu cosmic Release     
  404  Not Found [IP: 91.189.95.83 80]
Get:5 http://us.archive.ubuntu.com/ubuntu cosmic-updates InRelease [83.2 kB]
Hit:6 http://us.archive.ubuntu.com/ubuntu cosmic-backports InRelease
Reading package lists... Done                      
E: The repository 'http://ppa.launchpad.net/p-pisati/fuzix/ubuntu cosmic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

piso77, 2018-10-26:
Try it now, i should fixed it - there's only one package missing, the gcc6809, since it doesn't build on anything but Trusty.

jayacotton, 2018-10-27:
I think we can close this one now.   It works.

tnx
jc

piso77, 2018-10-27:
You can close it by clicking on the button.

closed 2018-10-27 by jayacotton