EtchedPixels/FUZIX #672: Problems creating bootable CF card.

jayacotton, 2018-11-07:
I'v been working on a bootable CF card for weeks now.  I am convinced that there is a bug somewhere
in the process.  I know that fuzix will boot on some systems, but the correct incantation is elusive for
rc2014.

What I get is "Disk Not Bootable!" with RomWBW.  

I have tried mkfs.cpm (no joy) and mkfs file type fat16 and cp/m (no joy)

These are the copy commands I have from email.

dd if=fuzix.romwbw of=/dev/sdX skip=2 
dd if=my.fs of=/dev/sdX1

jayacotton, 2018-11-10:
No joy.  I have not been able to get a bootable CF card.
What format command (mkfs) are use using to setup the card ?

jayacotton, 2018-11-14:
Should I be doing this, or using the linux version of mkfs ?

With this, fdisk does not recognize the card.

sudo ./mkfs /dev/sdc 512 65535
Making 512 byte/block filesystem with normal byte order on device /dev/sdc with fsize = 65535 and isize = 512.
Opening /dev/sdc (offset 0)

After sudo dd if=fuzix.romwbw of=/dev/sdc

Disk /dev/sdc: 65 KiB, 66560 bytes, 130 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x33479d11

and after sudo dd if=rootfs-z80-32 of=/dev/sdc1

Disk /dev/sdc1: 32 MiB, 33553920 bytes, 65535 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xde999ef0

These look o.k. to me, but I have a few questions.  In looking for what is happening, I find that romwbw
expects a sector size of 128 and the fuzix.romwbw file is formated with size of 128 but, the disk has a sector size of 512.  Is that a problem ?

jayacotton, 2018-11-20:
Still no joy.   I formatted the CF card on windows 10 (blanks the disk) and did the fdisk to create a partition
and did the dd command to load on the s/w all without a error messages.  Note: I can get the same results with the CF card not plugged in.

Still getting the "Disk not Bootable" .  When reading the romldr.asm file I see that it will only issue this
message if the signature is not present (or out of order). 

So, I down loaded a program called DU2.com and had a look at the disk blocks.  

Good news/Bad news.  I can see blocks and Identify the info that windows 10 format put on the disk.
However I can't find any of the boot loader information.   Looks like dd is not putting the file on the
CF card.  I am now wondering about the CF card programmer.  See above re error messages.

When I programmed the device I dd'd it on to the CF card then dd'd it off (using above commands) and
found that the results where the same.

So, maybe I am not dumping the correct track/sector on the CF card.  I expected it to be at 3 or 5 (128 byte) sectors.  Dumped from 0 to 20, found nothing.  Wondering about track numbers.
I did try track 1, nothing there.

jayacotton, 2018-11-20:
After a lot of head banging I now have this result:  
Please note the following.

Disk is still not bootable.  However, I have the boot loader located on the CF card.
It seems to be starting track 0 sector 10 (cp/m sectors i.e. 128) to about sector 5 (512 bytes).
This seems like I'm getting closer to figuring this thing out....

DU2  C0? r,d
00  33C08ED0 BC007C8E  C08ED8BE 007CBF00  *3@.P<.|.@.X>.|?.*
10  06B90002 FCF3A450  681C06CB FBB90400  *.9..|s$Ph..K{9..*
20  BDBE0780 7E00007C  0B0F850E 0183C510  *=>.....|......E.*
30  E2F1CD18 88560055  C6461105 C6461000  *bqM..V.UFF..FF..*
40  B441BBAA 55CD135D  720F81FB 55AA7509  *4A;*UM.]r..{U*u.*
50  F7C10100 7403FE46  10666080 7E100074  *wA..t..F.f`....t*
60  26666800 00000066  76086800 0068007C  *&fh....fv.h..h.|*
70  68010068 1000B442  8A56008B F4CD139F  *h..h..4B.V..tM..*

DU2  C0? s2
 Track = 0, Sector = 2, Physical Sector = 1

DU2  C0? r,d
00  83C4109E EB14B801  02BB007C 8A56008A  *.D..k.8..;.|.V..*
10  76018A4E 028A6E03  CD136661 731CFE4E  *v..N..n.M.fas..N*
20  11750C80 7E00800F  848A00B2 80EB8455  *.u.........2.k.U*
30  32E48A56 00CD135D  EB9E813E FE7D55AA  *2d.V.M.]k..>.}U**
40  756E7600 E88D0075  17FAB0D1 E664E883  *unv.h..u.z0Qfdh.*
50  00B0E660 E87C00B0  E664E875 00FBB800  *.0f`h|.0fdhu.{8.*
60  BBCD1A66 23C0753B  6681FB54 43504175  *;M.f#@u;f.{TCPAu*
70  3281F902 01722C66  6807BB00 00666800  *2.y..r,fh.;..fh.*

DU2  C0? s3
 Track = 0, Sector = 3, Physical Sector = 2

DU2  C0? r,d
00  02000066 68080000  00665366 53665566  *...fh....fSfSfUf*
10  68000000 00666800  7C000066 61680000  *h....fh.|..fah..*
20  07CD1A5A 32F6EA00  7C0000CD 18A0B707  *.M.Z2vj.|..M. 7.*
30  EB08A0B6 07EB03A0  B50732E4 0500078B  *k. 6.k. 5.2d....*
40  F0AC3C00 7409BB07  00B40ECD 10EBF2F4  *p,<.t.;..4.M.krt*
50  EB2BC9E4 64EB0024  02E0F824 02C3496E  *k+Iddk.$.`x$.CIn*
60  76616C69 64207061  72746974 696E2074  *valid partitin t*
70  61626C65 00457272  6F72206C 6F616469  *able.Error loadi*

DU2  C0? s4
 Track = 0, Sector = 4, Physical Sector = 3

DU2  C0? r,d
00  6E67206F 70657261  74696E67 20737973  *ng operating sys*
10  74656D00 4D697373  696E6720 70657261  *tem.Missing pera*
20  74696E67 20737973  74656D00 0000637B  *ting system...c{*
30  9A2CEB10 8D000000  20210007 FF000011  *.,k..... !......*
40  000E0723 00100000  00CC0100 00002021  *...#.....L.... !*
50  000CFEFF 00080000  0070DC01 00000000  *.........p\.....*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 55AAB2B2  *............U*22*

DU2  C0? s5
 Track = 0, Sector = 5, Physical Sector = 4

DU2  C0? r,d
00  00000000 00000000  00000000 00000000  *................*
10  00000000 00000000  00000000 00000000  *................*
20  00000000 00000000  00000000 00000000  *................*
30  00000000 00000000  00000000 00000000  *................*
40  00000000 00000000  00000000 00000000  *................*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 00000000  *................*

DU2  C0? s6,r,d
 Track = 0, Sector = 6, Physical Sector = 5
00  00000000 00000000  00000000 00000000  *................*
10  00000000 00000000  00000000 00000000  *................*
20  00000000 00000000  00000000 00000000  *................*
30  00000000 00000000  00000000 00000000  *................*
40  00000000 00000000  00000000 00000000  *................*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 00000000  *................*

DU2  C0? s7,r,d
 Track = 0, Sector = 7, Physical Sector = 6
00  00000000 00000000  00000000 00000000  *................*
10  00000000 00000000  00000000 00000000  *................*
20  00000000 00000000  00000000 00000000  *................*
30  00000000 00000000  00000000 00000000  *................*
40  00000000 00000000  00000000 00000000  *................*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 00000000  *................*

DU2  C0? s8,r,d
 Track = 0, Sector = 8, Physical Sector = 7
00  00000000 00000000  00000000 00000000  *................*
10  00000000 00000000  00000000 00000000  *................*
20  00000000 00000000  00000000 00000000  *................*
30  00000000 00000000  00000000 00000000  *................*
40  00000000 00000000  00000000 00000000  *................*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 00000000  *................*

DU2  C0? s9,r,d
 Track = 0, Sector = 9, Physical Sector = 8
00  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
10  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
20  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
30  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
40  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
50  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
60  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
70  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*

DU2  C0? s10,r,d
 Track = 0, Sector = 10, Physical Sector = 9
00  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
10  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
20  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
30  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
40  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
50  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
60  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
70  FFFFFFFF FFFFFFFF  5AA50000 00000000  *........Z%......*

jayacotton, 2018-11-20:
Formated with gparted
then sudo dd if=fuzix.romwbw of=/dev/sdc seek=2

As you can see below, I have the loader on the CF card, but it seems to be in the wrong place.

 Track = 0, Sector = 1, Physical Sector = 0

DU2  C0? r,d
00  FAB80010 8ED0BC00  B0B80000 8ED88EC0  *z8...P<.08...X.@*
10  FBBE007C BF0006B9  0002F3A4 EA210600  *{>.|?..9..s$j!..*
20  00BEBE07 3804750B  83C61081 FEFE0775  *.>>.8.u..F.....u*
30  F3EB16B4 02B001BB  007CB280 8A74018B  *sk.4.0.;.|2..t..*
40  4C02CD13 EA007C00  00EBFE00 00000000  *L.M.j.|..k......*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 00000000  *................*

DU2  C0? s2,r,d
 Track = 0, Sector = 2, Physical Sector = 1
00  00000000 00000000  00000000 00000000  *................*
10  00000000 00000000  00000000 00000000  *................*
20  00000000 00000000  00000000 00000000  *................*
30  00000000 00000000  00000000 00000000  *................*
40  00000000 00000000  00000000 00000000  *................*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 00000000  *................*

DU2  C0? s3,r,d
 Track = 0, Sector = 3, Physical Sector = 2
00  00000000 00000000  00000000 00000000  *................*
10  00000000 00000000  00000000 00000000  *................*
20  00000000 00000000  00000000 00000000  *................*
30  00000000 00000000  00000000 00000000  *................*
40  00000000 00000000  00000000 00000000  *................*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 00000000  *................*

DU2  C0? s4,r,d
 Track = 0, Sector = 4, Physical Sector = 3
00  00000000 00000000  00000000 00000000  *................*
10  00000000 00000000  00000000 00000000  *................*
20  00000000 00000000  00000000 00000000  *................*
30  00000000 00000000  D455ED66 00000020  *........TUmf... *
40  210083FE FF000800  000070DC 01000000  *!.........p\....*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 0055AAB2  *.............U*2*

DU2  C0? s5,r,d
 Track = 0, Sector = 5, Physical Sector = 4
00  00000000 00000000  00000000 00000000  *................*
10  00000000 00000000  00000000 00000000  *................*
20  00000000 00000000  00000000 00000000  *................*
30  00000000 00000000  00000000 00000000  *................*
40  00000000 00000000  00000000 00000000  *................*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
70  00000000 00000000  00000000 00000000  *................*

DU2  C0? s6,r,d
 Track = 0, Sector = 6, Physical Sector = 5
00  00000000 00000000  00000000 00000000  *................*
10  00000000 00000000  00000000 00000000  *................*
20  00000000 00000000  00000000 00000000  *................*
30  00000000 00000000  00000000 00000000  *................*
40  00000000 00000000  00000000 00000000  *................*
50  00000000 00000000  00000000 00000000  *................*
60  00000000 00000000  00000000 00000000  *................*
[... 31 lines cut ...]
50  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
60  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
70  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*

DU2  C0? s10,r,d
 Track = 0, Sector = 10, Physical Sector = 9
00  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
10  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
20  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
30  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
40  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
50  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
60  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
70  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*

DU2  C0? s11,r,d
 Track = 0, Sector = 11, Physical Sector = 10
00  FFFFFF5A A5000000  000000FF FFFFFFFF  *...Z%...........*
10  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
20  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
30  FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF  *................*
40  FFFFFFFF FF000000  00000000 46757A69  *............Fuzi*
50  78205742 57204C6F  61646572 24000000  *x WBW Loader$...*
60  F200F400 F2B2B2B2  B2B2B2B2 B2B2B2B2  *r.t.r22222222222*
70  B2B2B2B2 B2B2B2B2  B2B2B2B2 B2B2B2B2  *2222222222222222*

jayacotton, 2018-11-20:
Something I just noticed, the loader code is busted after the string "Fuzix WBW Loader" etc.

EtchedPixels, 2018-11-21:
The dd should be putting the loader on physical LBA 2 (disk LBA starts at 0 and in 512 sectors) - the tracks/sector stuff and 128 byte sectors is just an illusion created by CP/M.

You can use ROMWBW to read real disk sectors

https://github.com/wwarthen/RomWBW/blob/master/Doc/RomWBW%20Architecture.pdf

Something like

LD BC,#0x1230  ; comamnd 12 device 30 (IDE)
LD DE,0   ; block high
LD HL,2   ; block number
RST 8      ; call romwbw
LD BC,#0x1330              ; command
LD HL,#0x8000              ; somewhere above 32K
LD E,1                            ; 1 512 byte sector
RST 8

should read block 2

But all of this sounds like the windows dd isn't actually putting things in the right place on raw disk media as all that ROMWBW does for a disk boot is load LBA sector 2 off the disk, check it has the right marks and then loads/runs what it is told.

EtchedPixels, 2018-11-21:
If your kernel is small enough another thing that ought to work given the kernel image is loaded and run from 0x100 is to take the fuzix.bin it builds, transfer it into CP/M as FUZIX.COM and run it.

jayacotton, 2018-11-23:
I just got some new CF cards.  
sudo dd if=fuzix.romwbw of=/dev/sdc seek=2
boot with romwbw
"Reading disk information...
Boot failure!"

Clearly need to look for new errors.  But this looks promising ...

jayacotton, 2018-11-23:
Nope.  It looks like 16gig CF cards are not going to work.  Ordering some 128MB cards now.

jayacotton, 2018-11-24:
WRT kernel size.  It is to big to load and run.  I did try it, but got a load error.
This is the version of zcc that I am using:
zcc - Frontend for the z88dk Cross-C Compiler
v 20181116 (C) 1998-2016 D.J.Morris
Is there a 'better' version of the tool chain that will make smaller code ?

EtchedPixels, 2018-11-24:
I have no idea about zcc/z88dk fork of sdcc and Fuzix. I've never tried building it with that. I am currently using sdcc 3.8.1 for most builds.

jayacotton, 2018-11-25:
Still no joy on the kernel size.  the asbuilt fuzix.bin file is 58428 bytes.  

Inspecting the binary, I see that there is code up in the d200 range.  Don't know if that is a problem.

jayacotton, 2018-11-26:
Here is the last (most recent boot).  After this, I tried to repeat the same steps and now its back to 
disk not bootable.

This was done with Apacer 128mb flash card.  

There must be some step that I am ham fisting, I just don't know what it is.  

RetroBrew HBIOS v2.9.1-pre.3, 2018-05-11                                        
                                                                                
RC2014 @ 7.368MHz                                                               
0 MEM W/S, 1 I/O W/S, INT MODE 1                                                
512KB ROM, 512KB RAM                                                            
                                                                                
SIO0: IO=0x80 SIO MODE=115200,8,N,1                                             
SIO1: IO=0x82 SIO MODE=115200,8,N,1                                             
DSRTC: MODE=STD Thu 2000-01-27 00:12:17                                         
MD: UNITS=2 ROMDISK=384KB RAMDISK=384KB                                         
IDE: IO=0x10 DEVICES=1                                                          
IDE0: 8BIT LBA BLOCKS=0x0003D400 SIZE=122MB                                     
PPIDE: IO=0x20 NOT PRESENT                                                      
                                                                                
Unit        Device      Type              Capacity/Mode                         
----------  ----------  ----------------  --------------------                  
Disk 0      MD1:        RAM Disk          384KB,LBA                             
Disk 1      MD0:        ROM Disk          384KB,LBA                             
Disk 2      IDE0:       Hard Disk         122MB,LBA                             
Serial 0    SIO0:       RS-232            115200,8,N,1                          
Serial 1    SIO1:       RS-232            115200,8,N,1                          
                                                                                
RC2014 Boot Loader                                                              
                                                                                
Boot: (C)PM, (Z)System, (M)onitor,                                              
      (L)ist disks, or Disk Unit # ===> BOOT FROM DISK                          
                                                                                
Reading disk information...                                                     
                                                                                
LOC=F200 END=F400 ENT=F200 LABEL=Fuzix WBW Loader                               
Loading...                                                                      
IDE0: COMMAND ERROR                                                             
IDE0: COMMAND ERROR                                                             
IDE0: COMMAND ERROR                                                             
IDE0: COMMAND ERROR�

jayacotton, 2018-12-08:
I changed over to a spinning hard drive.  That has helped a lot.
I found that the signature bytes are reversed. A simple fix with hexedit got me past that part.
Now I can boot the kernel with some consistency.

So, I will close this one and start a new one for sio problems.

closed 2018-12-08 by jayacotton

jayacotton, 2018-12-09:
After updating to the very latest code tree, I no longer need to swap the signature bytes.

jayacotton, 2018-12-12:
So, after testing 20 different CF cards, I found 32gb sandisk works.  But you have run gparted on the CF card before loading updates to the card.

I have only CF card configured and this is satisfactory for now.