EtchedPixels/FUZIX #676: Serial port is wacked !

jayacotton, 2018-12-08:
See below.  The first part is the romwbw text messages etc.

RetroBrew HBIOS v2.9.1-pre.3, 2018-05-11

RC2014 @ 7.368MHz
0 MEM W/S, 1 I/O W/S, INT MODE 1
512KB ROM, 512KB RAM

SIO0: IO=0x80 SIO MODE=115200,8,N,1
SIO1: IO=0x82 SIO MODE=115200,8,N,1
DSRTC: MODE=STD Tue 2000-02-08 00:12:20
MD: UNITS=2 ROMDISK=384KB RAMDISK=384KB
IDE: IO=0x10 DEVICES=1
IDE0: 8BIT LBA BLOCKS=0x0003E800 SIZE=125MB
PPIDE: IO=0x20 DEVICES=2
PPIDE0: LBA BLOCKS=0x037EAC80 SIZE=28629MB
PPIDE1: NO MEDIA

Unit        Device      Type              Capacity/Mode
----------  ----------  ----------------  --------------------
Disk 0      MD1:        RAM Disk          384KB,LBA
Disk 1      MD0:        ROM Disk          384KB,LBA
Disk 2      IDE0:       Hard Disk         125MB,LBA
Disk 3      PPIDE0:     Hard Disk         28629MB,LBA
Disk 4      PPIDE1:     Hard Disk         --
Serial 0    SIO0:       RS-232            115200,8,N,1
Serial 1    SIO1:       RS-232            115200,8,N,1

RC2014 Boot Loader

Boot: (C)PM, (Z)System, (M)onitor,
      (L)ist disks, or Disk Unit # ===> BOOT FROM DISK

Reading disk information...

LOC=F200 END=F400 ENT=F200 LABEL=Fuzix WBW Loader

<-------------------------- kernel booted and started here ------------------------------->

evooht(c 214208 AanCo <la@echdpxes.o.k>m> a una
512kBtoalRA, 432kBavilbl t pocsss (15 pocsss ax
Enbin nerut ..ok
IDEdrve 0: hda: 
botev  1: ideeror sats=0000

Something very odd with the sio data.  I did try replacing the sio/2 board, no difference.  I also tried the
vga terminal vs the sio/usb line on my computer.   Same picture.

It looks like its missing every other character, also when I type into the terminal it ignores the first 
character.

EtchedPixels, 2018-12-09:
It's misdetecting the SIO v ACIA. The end result is it thinks the SIO is always ready for another byte so lots of them vanish.

Weird I thought the latest code I'd nailed that one down for good.

jayacotton, 2018-12-09:
The very latest source code still shows the bug.

c32ff0, 2018-12-09:
while working on a similar port i had to add some similar code to the following at the start of tty_putc() on Kernel/platform-rc2014/devtty.c :
	__asm
	poll:          ; poll the device for remaining characters
    	sub a,a        ; zero out A
    	out (0x80),a   ; select read registart 0
    	in a,(0x80)    ; Read it 
    	bit 2,a
    	jr z,poll     ; if carry is 0 ( character still being sent ) jump to poll
	__endasm;

maybe you can try that but it's most likely something else
(for the Scott Baker SIO card, change the 0x80 to 0x82)

c32ff0, 2018-12-09:
quick fix: add this code at the start of tty_putc() on Kernel/platform-rc2014/devtty.c :
__asm
poll: ; poll the device for remaining characters
sub a,a ; zero out A
out (0x80),a ; select read registart 0
in a,(0x80) ; Read it
bit 2,a
jr z,poll ; if carry is 0 ( character still being sent ) jump to poll
__endasm;
for the Scott Baker SIO card, change the 0x80 to 0x82
Also this assumes you only use port A

EtchedPixels, 2018-12-09:
rc2014 does have it working if it detects right - the tty_writeready() function handles it. It's just that if it thinks you have an ACIA it ends up polling bit 1 of the ACIA control port and then crazy happens.

Easiest hack to just disable the probe is to edit rc2014.s and find init_hardware

After the out (FDC_DOR),a stick a jp is_iso

and it'll blindly assume you have an SIO.

I'll take a deeper look at this when I'm next round to working on RC2014 based machines.

jayacotton, 2018-12-09:
I put in both fixes, and it works !  thanks.

EtchedPixels, 2018-12-10:
I've pushed what I hope is an actual fix using the RomWBW algorithm.

jayacotton, 2018-12-11:
Bad news.  The fix does not work (for me)...  Seems to be the same bug.

jayacotton, 2018-12-16:
I have not tried an acia board yet.  I'll run that test and post the results.

jayacotton, 2018-12-16:
Highly busted also.

EtchedPixels, 2019-02-01:
Now seems to be working

closed 2019-02-12 by EtchedPixels