EtchedPixels/FUZIX #680: Help: stuck in start.c and can't get out.

jayacotton, 2018-12-10:
After booting the startup code (and kernel I guess) from the ppide hard drive, I need to enter the 
drive (name/nr) to mount for the root file system.

This is where things get odd.  The CF card driver identifies and enumerates drives,
however the PPIDE driver does not.  Hence I don't know that the name of the drive is supposed to be.

I did try to guess, but after 3 hrs of tinkering and reading code, I have to admit defeat.

RetroBrew HBIOS v2.9.1-pre.3, 2018-05-11

RC2014 @ 7.376MHz
0 MEM W/S, 1 I/O W/S, INT MODE 1
512KB ROM, 512KB RAM

SIO0: IO=0x80 SIO MODE=115200,8,N,1
SIO1: IO=0x82 SIO MODE=115200,8,N,1
DSRTC: MODE=STD Wed 2000-02-09 07:11:48
MD: UNITS=2 ROMDISK=384KB RAMDISK=384KB
IDE: IO=0x10 DEVICES=1
IDE0: 8BIT LBA BLOCKS=0x0003E800 SIZE=125MB
PPIDE: IO=0x20 DEVICES=2
PPIDE0: LBA BLOCKS=0x037EAC80 SIZE=28629MB
PPIDE1: NO MEDIA

Unit        Device      Type              Capacity/Mode
----------  ----------  ----------------  --------------------
Disk 0      MD1:        RAM Disk          384KB,LBA
Disk 1      MD0:        ROM Disk          384KB,LBA
Disk 2      IDE0:       Hard Disk         125MB,LBA
Disk 3      PPIDE0:     Hard Disk         28629MB,LBA
Disk 4      PPIDE1:     Hard Disk         --
Serial 0    SIO0:       RS-232            115200,8,N,1
Serial 1    SIO1:       RS-232            115200,8,N,1

RC2014 Boot Loader

Boot: (C)PM, (Z)System, (M)onitor,
      (L)ist disks, or Disk Unit # ===> BOOT FROM DISK

Reading disk information...

LOC=F200 END=F400 ENT=F200 LABEL=Fuzix WBW Loader
Loading...FUZIX version 0.3pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2018 Alan Cox <alan@etchedpixels.co.uk>
Devboot
512kB total RAM, 432kB available to processes (15 processes max)
Enabling interrupts ... ok.
IDE drive 0: hda: 
IDE drive 1: ide error, status=0000
bootdev:

jayacotton, 2018-12-10:
At least part of the problem is that Kernel/platform-rc2014/discard.c (discard.lst) does not compile the
ppide_init call.  I will force it to compile and see what happens next.

beretta42, 2018-12-10:
I'm not total sure on rc2014 internals (I'm a CoCo man), but the "bootdev" prompt takes a device number. (it can also take a symbolic name).  I think a simple '0' will mount root from your IDE drive.  This major (0) points to the blkdev devices, which further breaks out the number into drives and partitions.

see /Kernel/dev/blkdev.c  for an explanation of how it treats the numbers.....

beretta42, 2018-12-10:
modify rc2014's "config.h".   modify  
``` #undef CONFIG_RC2014_PPIDE``` 
to read: 
```#define CONFIG_RC2014_PPIDE```
If I follow the code correctly that should make discard.c call ppide_init().

jayacotton, 2018-12-10:
Yes, that did get the call to ppide_init() into discard.rel   
Still getting no traction.  
With the current config.h file and a fresh build of all the source in Kernel/...
I get:
Reading disk information...

LOC=F200 END=F400 ENT=F200 LABEL=Fuzix WBW Loader
Loading...FUZIX version 0.3pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2018 Alan Cox <alan@etchedpixels.co.uk>
Devboot
512kB total RAM, 432kB available to processes (15 processes max)
Enabling interrupts ... ok.
IDE drive 0: ide timeout, status=0060
IDE drive 1: ide timeout, status=0060

The timeouts are very long (order 60 seconds) and the spinning rust is not on the list, nor can I 
just punch in numbers... 0 to 16 gets things like "Mounting root fs (root_dev=3, rw): failed"

Still think there is a deeper problem here.....

EtchedPixels, 2018-12-10:
I have not yet tested the PPIDE version of the set up on the RC2014 hardware. I'm still building my PPIDE card. I will double check the register setups as it should 'just work' because it's the same as the PPIDE hardware on SBCv2 and similar systems.

jayacotton, 2018-12-11:
There is a very elaborate bit of code for CF cards in ~/dev/devide_discard.c.  This code seems to plumb the disk device.  No such function exists for ppide.  

Should ppide get plumbed in like devide?  perhaps expanding devide_discard.c to include ppide devices ?

EtchedPixels, 2018-12-11:
PPIDE uses the same devide_discard.c code. The low level IDE routines change but the high level code is the same for all IDE/ATA interfaces.

jayacotton, 2018-12-16:
turns out that even if I boot from a CF card with the ppide card present, I can't move to mount the root fs.

jayacotton, 2018-12-17:
New info.  I put a CF card to IDE interface on my IDE board.  I can boot from it, but I can't see it after
booting from it.  All I can see is the rc2014-CF card interface.

jayacotton, 2019-01-10:
The one bit change to platform_ide.h did not fix the problem.  I will now start to dig in to this code and see if I can find a bug.

jayacotton, 2019-01-10:
I did not quite get that right.  It now presents a different message.
Reading disk information...

LOC=F200 END=F400 ENT=F200 LABEL=Fuzix WBW Loader
Loading...FUZIX version 0.3pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2018 Alan Cox <alan@etchedpixels.co.uk>
Devboot
512kB total RAM, 432kB available to processes (15 processes max)
Enabling interrupts ... ok.
IDE drive 0: hda: hda1 hda2 hda3 hda4 
IDE drive 1: ide error, status=0000
bootdev: hda1
Mounting root fs (root_dev=1, ro): failed
bootdev: hda1
Mounting root fs (root_dev=1, rw): failed
bootdev: 

It seems to be plumbing in the disk, but its not allowing a mount.  More IDE address problems ?

EtchedPixels, 2019-01-10:
It read the partition table correctly so the driver is working
It then tried to read the first block of the filesystem on sda1 and it wasn't a fuzix one

Looks like it's working to me

What does the partition table look like if you dump with it fdisk or similar tools. In particular what is the LBA offset of the first partition - is it 63 or 2048 ?

If it's 63 and you added the kernel after the file system then that would make sense because the loaders assume a 21st century disk partitioning scheme where there are more than 32K of boot space.

closed 2019-02-16 by EtchedPixels