EtchedPixels/FUZIX #698: trying to build for z80pack

jcw, 2019-04-12:
Hello - I'd like to build FUZIX to explore and tinker with it. Decided to start with a `z80pack` build before moving on to actual hardware. On Ubuntu 18.04 w/ sdcc 3.80 (binary amd64 install from sourceforge), I get the following error:

```
yacc  find.y
mv -f y.tab.c find.c
../../../Library/tools/fcc -mz80   -O2 -c find.c
y.tab.c:1233: warning 112: function 'yylex' implicit declaration
y.tab.c:1498: warning 112: function 'yyerror' implicit declaration
y.tab.c:1642: warning 112: function 'yyerror' implicit declaration
y.tab.c:1498: error 101: too many parameters
y.tab.c:1642: error 101: too many parameters
find.y:216: warning 85: in function yyerror unreferenced function argument : 'p'
```

Could this be due to using an unsuitable version of `sdcc` - and if so, which version would be best? I see that there's a very recent `sdcc-3.9.0-rc1` out ...

-jcw

EtchedPixels, 2019-04-12:
The yacc one is an Ubuntu/Debian packaging bug. They by default make yacc run bison which is sort of yacc but not enough.

See INSTALL.  I'll update the last note to mention Ubuntu and other Debian derivatives have the same fix needed

closed 2019-04-12 by EtchedPixels

jcw, 2019-04-12:
_Perfect fix_ - many thanks!