EtchedPixels/FUZIX #699: trying to debug a new z80 port, some Q's

jcw, 2019-04-14:
Hello - I'm trying to debug a new port to a custom Z80 emulator ([link](https://jeelabs.org/2018/z80-zexall-bp/)) - my code is different from z80pack, in that it supports setting the starting address of the common bank (set to 0xF000 on init), as well as setting/getting the bank (currently 8 banks, with bank 0 used for the kernel).

What I'm getting so far is:

```
FUZIX version 0.3pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2019 Alan Cox <alan@etchedpixels.co.uk>
Devboot
484kB total RAM, 420kB available to processes (15 processes max)
Enabling interrupts ... ok.
Mounting root fs (root_dev=256, ro): HD wr 0 mem 0xAD43 pos 1
HD wr 0 mem 0xAB3B pos 2
OK
Starting /init
HD wr 0 mem 0xA933 pos 256
HD wr 0 mem 0xA72B pos 19
HD wr 0 mem 0xA523 pos 294
```

Those "HD ..." lines are disk access debug prints.

The difference with z80pack's banked memory, seems to be that I don't really have a way to turn banking off, once enabled, nor write-protect anything - I can merely switch between banks 0..7.

Evidently, the logic in `z80pack.s` needs to be adjusted to this, but I haven't figured out yet how I/O on port 23 should be handled: can I simply comment out those protect/unprotect calls?

Also, what sort of disk I/O should I expect as it's starting `init` ? I see some bank switches between banks 0 and 7, but I assume the next thing that should happen is reads into bank 7?

(FWIW: many things are not yet implemented, such as no interrupts or tty input yet)

EtchedPixels, 2019-04-14:
I would expect to see a tiny amount of kernel targetted I/O mounting which you have. It then should read the path and inode to find /init and then load the first block of init into a kernel buffer, and the rest directly into user space if you have direct I/O enabled.

You may find the sbcv2 port is much better as a guide. It's a bit different in split (32K/32K) but heavily commented and has no weirdness like memory protection.

What I normally do at that point in debug is to log the various stages through _execve() as it tells you a lot about where it is dying and what stage of the execution you are at.

jcw, 2019-04-14:
Ah, thanks - I now see that the `mmu_*` stuff is optional. That sure simplifies things. Progress!

jcw, 2019-04-14:
Ok, the init greeting shows, am now running into an infinite `fork` loop. Could it be that I need to get the timer interrupts going at this point? Or can the kernel task-switch without interrupts, somehow?

I'll re-read the notes about "tickless".

EtchedPixels, 2019-04-14:
The init version ... message was printed by init so that is a good sign that init loaded correctly as it was able to print a message from its data segment, which means it called into the kernel and did a syscall properly then by the looks of it tried to fork /etc/rc and in turn /bin/sh

If you have enough free space there is a define in process.c that turns on syscall tracing and can be very informative.

jcw, 2019-04-14:
Ooh, brilliant - I've got something to chew on:

```
                pid 1: syscall 8        write(0001, 1083, 0017)
init version 0.9.0ac#1
                        pid 1: ret syscall 8, ret 0017 err 0000
                pid 1: syscall 31       sbrk(0000, 006C, F5F2)
                        pid 1: ret syscall 31, ret 4139 err 0000
                pid 1: syscall 1        open(0B75, 0000, 0028)
                        pid 1: ret syscall 1, ret 0003 err 0000
                pid 1: syscall 16       _fstat(0003, EDB6, EDB6)
                        pid 1: ret syscall 16, ret 0000 err 0000
                pid 1: syscall 31       sbrk(0113, 0028, 0028)
                        pid 1: ret syscall 31, ret 4139 err 0000
                pid 1: syscall 7        read(0003, 4139, 0112)
HD wr 0 mem 0:0xA82E pos 286
                        pid 1: ret syscall 7, ret 0112 err 0000
                pid 1: syscall 2        close(0003, 0112, 0000)
                        pid 1: ret syscall 2, ret 0000 err 0000
                pid 1: syscall 30       brk(41C6, 09BD, 41C6)
                        pid 1: ret syscall 30, ret 0000 err 0000
                pid 1: syscall 32       _fork(0000, 0000, 028D)
fork
                        pid 2: ret syscall 32, ret 0000 err 0000
                pid 2: syscall 39       kill(0000, 0007, 0000)
                        pid 2: ret syscall 39, ret 0000 err 0000
                        pid 1: ret syscall 32, ret 0002 err 0000
                pid 1: syscall 55       waitpid(0002, 0000, 0000)
                        pid 1: ret syscall 55, ret 0002 err 0000
                pid 1: syscall 1        open(345D, 0000, 0000)
                        pid 1: ret syscall 1, ret FFFF err 0002
                pid 1: syscall 32       _fork(0000, 0000, 11FA)
fork
                        pid 3: ret syscall 32, ret 0000 err 0000
                pid 3: syscall 39       kill(0000, 0007, 0000)
                        pid 3: ret syscall 39, ret 0000 err 0000
                        pid 1: ret syscall 32, ret 0003 err 0000
                pid 1: syscall 55       waitpid(FFFF, 0000, 0000)
                        pid 1: ret syscall 55, ret 0003 err 0000
                pid 1: syscall 1        open(1F38, 0002, 0002)
HD wr 0 mem 0:0xA626 pos 7
HD wr 0 mem 0:0xAA36 pos 263
HD wr 0 mem 0:0xAE46 pos 276
                        pid 1: ret syscall 1, ret FFFF err 0002
                pid 1: syscall 1        open(345D, 0000, 0000)
                        pid 1: ret syscall 1, ret FFFF err 0002
                pid 1: syscall 32       _fork(0000, 0000, 11FA)
fork
                        pid 4: ret syscall 32, ret 0000 err 0000
                pid 4: syscall 7        read(0000, 0000, 0000)
                        pid 4: ret syscall 7, ret 0000 err 0000
                pid 4: syscall 39       kill(0000, 0007, 0007)
                        pid 4: ret syscall 39, ret 0000 err 0000
                        pid 1: ret syscall 32, ret 0004 err 0000
                pid 1: syscall 55       waitpid(FFFF, 0000, 0000)
                        pid 1: ret syscall 55, ret 0004 err 0000
                pid 1: syscall 1        open(1F38, 0002, 0002)
                        pid 1: ret syscall 1, ret FFFF err 0002
                pid 1: syscall 1        open(345D, 0000, 0000)
                        pid 1: ret syscall 1, ret FFFF err 0002
                pid 1: syscall 32       _fork(0000, 0000, 11FA)
fork
                        pid 5: ret syscall 32, ret 0000 err 0000
                pid 5: syscall 39       kill(0000, 0007, 0007)
                        pid 5: ret syscall 39, ret 0000 err 0000
                        pid 1: ret syscall 32, ret 0005 err 0000
                pid 1: syscall 55       waitpid(FFFF, 0000, 0000)
                        pid 1: ret syscall 55, ret 0005 err 0000
                pid 1: syscall 1        open(1F38, 0002, 0002)
                        pid 1: ret syscall 1, ret FFFF err 0002
                pid 1: syscall 1        open(345D, 0000, 0000)
                        pid 1: ret syscall 1, ret FFFF err 0002
                pid 1: syscall 32       _fork(0000, 0000, 11FA)
fork
[...]
```

Many thanks, again. This is indeed going to help a lot - to be continued ...

beretta42, 2019-04-15:
Great progress in 9 hours!  kputs() and kprintf() are your friends too.

jcw, 2019-04-15:
Thanks, the Z80 emulator I'm using for this is in [this repository](https://git.jeelabs.org/jcw/retro/src/branch/master/native-fuzix), but my FUZIX hacking is not online yet.

EtchedPixels, 2019-04-15:
Your /bin/sh is failing. What compiler/tools did you build the image with ?

Try rebuilding /bin/sh with 3.9.0-rc1 and dropping that into your file system, also try the /bin/sh from a 0.2 image off www.fuzix.org ?

jcw, 2019-04-15:
Ok - will do. Not sure which image I'm currently using (is "init version 0.9.0ac#1" a hint?), I _think_ I did a rebuild using 0.3rc1, after the yacc fix. FWIW, I ran into the same yacc/bison issue on MacOS (homebrew only offers bison), so I'll need to switch back to Linux for that. Not a show-stopper, but might it be an idea to include the yacc output in the repository?

EtchedPixels, 2019-04-15:
What matters is the sdcc version. late sdcc 3.8.x miscompile /bin/sh. Also the older tools for 0.3-rc1 or so misbuild /bin/sh in other ways that then broke the current kernel.

jcw, 2019-04-15:
Yes, understand. I'm using the latest (`brew install --HEAD sdcc`), as far as I can tell:

```
$ sdcc -v
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ez80_z80/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8/pdk14/pdk15 3.9.0 #11195 (Mac OS X x86_64)
published under GNU General Public License (GPL)
```

EtchedPixels, 2019-04-15:
Known good binary attached

[sh.gz](https://github.com/EtchedPixels/FUZIX/files/3081019/sh.gz)

jcw, 2019-04-15:
Thank you! I've copied it in, but the fork runaway remains. I suspect that I'm still not handling all the bank switching right - maybe messing up A or HL or some other reg. FWIW, I've attached my hacked `z80pack.s` in case you have time to spare to check it out. There are some important notes at the start.

I'd expect the `fork` to continue with an `exec`, but the output is unchanged. I don't see any substantial reads of the `sh` binary, nor my `execve 1` debug msg at the start of `execve()`. It's dying before that.

[z80pack.s.gz](https://github.com/EtchedPixels/FUZIX/files/3081098/z80pack.s.gz)

jcw, 2019-04-15:
My conclusion so far (please let me know if I'm being too chatty...) is that child processes never run. The first fork is `/etc/rc` and the /bin/sh for it is never exec'ed, since only init runs, i.e. the original parent (which explains why all sh variants and date never showed up). All the subsequent fork's are internal getty calls, but again the child code in init is never executed.

jcw, 2019-04-15:
Bingo. A register got clobbered:

```
map_process_a:
            push af ; ADDED
	    in a, (7) ; CHANGES A !
            pop af ; ADDED
            ret
```

Now it all works as expected:

```
FUZIX version 0.3pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2019 Alan Cox <alan@etchedpixels.co.uk>
Devboot
484kB total RAM, 420kB available to processes (15 processes max)
Enabling interrupts ... ok.
Mounting root fs (root_dev=256, ro): OK
Starting /init
init version 0.9.0ac#1
Checking root file system.
```

And then it quits in `platform_rtc_read`, which isn't being emulated yet. Onwards!

jcw, 2019-04-16:
All is well, with proper tty input logic:

```
FUZIX version 0.3pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2019 Alan Cox <alan@etchedpixels.co.uk>
Devboot
484kB total RAM, 420kB available to processes (15 processes max)
Enabling interrupts ... ok.
Mounting root fs (root_dev=256, ro): OK
Starting /init
init version 0.9.0ac#1
Checking root file system.
Current date is Thu 1970-01-01
Enter new date: 2019-04-16
Current time is  0:00:00
Enter new time: 07:18:00

 ^ ^
 n n   Fuzix 0.3rc1
 >@<
       Welcome to Fuzix
 m m

login: root

Welcome to FUZIX.
# ls -l /
drwxr-xr-x   2 root     root         3584 Apr 16 00:21 bin
drwxr-xr-x   2 root     root         3072 Apr 16 00:21 dev
drwxr-xr-x   2 root     root          512 Apr 16 00:21 etc
-rwxr-xr-x   1 root     root        16668 Apr 16 00:21 init
drwxr-xr-x   2 root     root          512 Apr 16 00:21 mnt
drwx--x---   2 root     root          512 Apr 16 00:21 root
drwxrwxrwt   2 root     root          512 Apr 16 00:21 tmp
drwxr-xr-x  10 root     root          512 Apr 16 00:21 usr
drwxr-xr-x   5 root     root          512 Apr 16 00:21 var
# df
Filesystem       Blocks   Used   Free  %Used Mounted on
/dev/fd0          16384   5740  10388    35% /
# ps
  PID    TTY     TIME CMD
    9    tty1 00:00:00 -sh
   14    tty1 00:00:00 ps
#
```

Now the fun tweaking starts :)
Many thanks for all the help, I'm closing this issue.

beretta42, 2019-04-16:
Huzzah!

jcw, 2019-04-16:
Heh - thanks. But the point of this exercise was really to get this same emulator running off an SD card on [this µC board](https://www.ebay.com/itm/182309828654), which can support 3x 60K + 4K common. I can confirm that it actually works if I hack the banking a bit, but 2 user banks require proper swap support, and that ain't quite working yet ...

Sooo... I'm closing on on running FUZIX on €10 hardware, but it's not _quite_ there yet :)

jcw, 2019-04-18:
Sooo close ...

```
--- Miniterm on /dev/cu.usbmodemE0C2C5A73  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

sd init: detected, hd=0
booting: ...............................................................................................................................
FUZIX version 0.3pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2019 Alan Cox <alan@etchedpixels.co.uk>
Devboot
184kB total RAM, 120kB available to processes (15 processes max)
Enabling interrupts ... ok.
Mounting root fs (root_dev=256, ro): OK
Starting /init
init version 0.9.0ac#1
Checking root file system.
/etc/rc: setdate: not found
50C9_switchin: FAIL
```

Looks like µSD card I/O is not quite right yet.

jcw, 2019-04-21:
(re-opening, I hope that's ok)

My port now runs properly on MacOS, but not on ARM (STM32). Here's a shorthand of what's happening - the character pairs `r<N>` and `w<N>` are reading and writing to bank `<N>`, respectively:

```
sd init: detected, hd=0
booting: .......................................................................
........................................................
FUZIX version 0.3pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2019 Alan Cox <alan@etchedpixels.co.uk>
Devboot
184kB total RAM, 120kB available to processes (15 processes max)
Enabling interrupts ... ok.
Mounting root fs (root_dev=256, ro): r0r0OK
Starting /init
r0r0r0r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r0r2r2r2r2r2r2r2r2r2r2r2r2r2r2r0r0r0r0r0
r0init version 0.9.0ac#1
r0r0r0r0r0r0r0r0r0r0r0r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r0r1r1r1r1r1r1r1r1r1r1r1
r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r0W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2
W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2
W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2
W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2r0r0r0r0r0W1W1W1W1W1W1W1W1W1W1W1
W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1
W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1
W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1r0r0r0r0r0r1r1r1r1r1r1
r1r1r1r1r1r1r1r1r1r1r1r0r1r1r1r1r1r1r1r1r1r0r0r0r0r0r0r1r0r0r0r0r1r0r0r0r1r0r0r0
r1r0r0r0r1r0r0r0r0r0r0r0r0r0r0r0r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r0r1r1r1r1r1r1
r1r1r1r1r1r1r1r1r1r1r0r0r0r0r0r0r0r0Checking root file system.
r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1
r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1
r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1
r1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1
W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1
W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1W1
W1r0r0r0r0r0r0r0r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r1r0r1r1r1r1r1r1r1r1r1r1r1r1r1r1
r1r1r1r1r1r1r1r1r1r1r1r0r0r0r0r0r0r0r0r0r0Current date is Thu 1970-01-01
Enter new date: W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2
W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2
W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2W2
W2W2W2W2W2W2W2W2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2
r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2
r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2r2
r2r2r2r2r2r2r2r250C4_switchin: FAIL
```

This is running with 3 banks of 60K (2 user) and 4K common, as before. So it's clear that `init` gets loaded in the last-added bank, i.e. bank 2. All the execs seem to be ok, i.e. `sh`, `mkfs`, and `setdate`.

The kernel and rootfs images are identical for macos and f407, and I'm running the same z80emu emulator. So the only difference should be in the way I interpret emulator calls for conin, conout, disk I/O, etc. But the strange thing to me, is that the `Enter new date:` prompt is not waiting for input.

Sorry for rambling a bit - I'm somewhat out of ideas on how to chase this further at the moment.

PS. I've pushed my code and reorganised my repository a bit - the emulator main for ARM can now be found at <https://git.jeelabs.org/jcw/retro/src/branch/master/z80emu/fuzix-arm/src/main.cpp>.

jcw, 2019-04-22:
My bad. The SD card driver timing was wrong, writes were unreliable.
Closing for good this time. With apologies for the noise.

closed 2019-04-22 by jcw

jcw, 2019-04-24:
Just to finish on a fun note - FUZIX now works properly via `z80emu` on the following platforms:

* ARM Cortex M4 @ 168 MHz - i.e. STM32F407 w/ 192K ram, plus a µSD card
* ESP32 @ 240 MHz - i.e. a WROVER module w/ 4M psram, plus a µSD card
* native on MacOS (well ... emulated, so not _really_ native)

All of these run the same kernel image (and root FS) - 64K main mem, with two banks of 60K. Swapping to SD is fairly slow at the moment, and shutdown sometimes leaves the FS in a bad state which `fsck` can't properly recover from (then again, I haven't implemented s/w restart).

Memory mapping: 64K and 120K are in two different RAM areas on F407, on ESP32 it gets messy, so the 120K for banks is allocated in 4K chunks (the heap is already too fragmented, even at startup, for a large single alloc), and on the MacOS laptop it's just one big array in `bss`, of course.

Missing features: no interrupts or RTC support yet. Fixed banking uses the `bouncebuffer` in common, which can probably be improved. That would reduce the common area, allowing 2K iso 4K common.

A few observations w.r.t. this setup:

* these ports are all very early and full of nasty code (and no doubt some bugs) - "proof of concept"
* my impression is that even the ARM and ESP32 emulations are snappier than a real Z80 @ 4 MHz
* swaps transfer the full 60K bank, it would be nice if this could be reduced via some sort of high-water mark, and perhaps there is a r/o code portion which need not be swapped out (or not again?)
* I can't help but wonder whether FUZIX will ever evolve to be able to do self-hosted compiles ...

_Anyway - it's amazing to see a Z80 run in such a "unix'y" fashion (even if emulated) !_

EtchedPixels, 2019-04-25:
shutdown leaving the fs mashed sounds like something bad is happening. Even with a regular hard off when testing I don't see that problem on other ports.

z80pack on a 3GHz core based x86 gives you about a 2GHz Z80 so that wouldn't surprise me.

For swap if you look at the logic in simple.c and bankfixed.c it does a single write out. There are two unrelated improvements
- Only swap what you need
- Allocate space for only what you need

The second is hard the first is easy. If you want to see what difference it makes you need to hack it to do two writes one from base -> udata.u_break the other from brk_limit() to p->p_top, where you round the two up and down to get block sized chunks.

simple swaps in the udata first, so should be easy to tweak. bankfixed is much more complicated because the udata is read as part of the swapin so you don't know u_break (or brk_limit()) when you are doing the swapin. That would require rather more reworking.

How big a difference it makes will depend a lot on the system and I/O speed. Probably quite a lot if your I/O is slow.