EtchedPixels/FUZIX #700: direct I/O & swap Q's

jcw, 2019-04-16:
(me again) - I'm trying to debug some I/O issues. Two problems I'm running into:

* direct I/O is not working -  it hangs after reading what looks like the last block of `init` in raw=0, right after all the raw=2 2 calls, which I can see are going to the right address and bank

I'm running without direct I/O for now. The second puzzle, which I think is more urgent, is:

* swapouts seem to be ok - 120 blocks are written to the swap device, but swapin fails:

```
hd0: W @ 1	#1 addr 0:AC46 desc 0xC28F raw 0
Swapping in B4F8 (page 4), utab.ptab B541
hd1: r @ 7320	#120 addr 4:0000 desc 0xC28F raw 2
Swapped in B4F8 (page 4), udata.ptab B541
0049_switchin: FAIL
```

What I see is that it gets swapped in to a different bank, but that should be fine IMO.

The way I'm running the swap test is to run nested shells until I see a swapout, then exit back from each shell until a swapin is required. That immediately runs into the above issue.

I'll read through some more of the code and docs, but if anyone has tips on how to chase this ...

EtchedPixels, 2019-04-16:
Where is your udata stash located ?

I would also dump the relevant bits of the disk image after the swap and see if what you have is valid. There is a lot (almost everything) in common between the direct I/O and swap I/O path in banked mode so they may be connected.

The switchin error occurs when you swap in a process or restore its data and the udata.ptab is not the right one. That often means your swapin reads actually went nowhere or the wrong place - which would also fit your direct I/O fail

jcw, 2019-04-16:
Ok - U_DATA_STASH = 0xEE00 in kernel.def, same as PROGTOP in config.h - MA_SIZE is 0xF000.

I'll check that the process data itself is actuall y going to the swap disk and being read back in. That would indeed explain what's going in.

jcw, 2019-04-16:
Found it. Was passing some data into to the disk driver via a struct that got "banked away". Solved by placing the info on the stack instead. Doh (in hindsight).

And indeed, this also fixed the direct I/O bug.

Thanks for pointing me in the right direction. Checking the actual I/O activity did the trick!

EtchedPixels, 2019-04-16:
Been there, done that with variables a few times

jcw, 2019-04-17:
Oops, not there yet - the target I'm aiming for has 184 KB of (emulated) RAM, mapped as 3 banks of 60K + 4K common. When I lower MAX_MAPS to 2 without changing anything else, the boot chokes after the init greeting - no disk requests, only a bunch of bank switches between kernel and user mode.

Do I need to make further changes? I think the problems start at MAX_MAPS of 5 and below.

EtchedPixels, 2019-04-17:
1 map won't work with banked (use simple) but 2 maps certainly should. Testing it on the 8085 emulator I have open right now it works there.

Did you make clean before rebulding ?

jcw, 2019-04-18:
Thanks for checking. Yes, I did a `make kclean`, but I'll retrace my steps and double-check.

jcw, 2019-04-18:
I don't have a working RTC or interrupts yet. Could that be the problem? Ok with 6, fails with 5 banks:

```
HD0 wr 0 mem 0:0xA0FC pos 280
HD0 wr 0 mem 0:0xA50C pos 281
HD0 wr 0 mem 0:0xAD2C pos 20
Current date is Thu 1970-01-01
Enter new date: 0049_switchin: FAIL 1
```

EtchedPixels, 2019-04-18:
RTC and interrupts won't cause swap problems. Wrong sizes causing overlapping writes migh

The 0049 looks like the size of a process so again as if the swapin failed or got the wrong thing. You can trace swap with the debug lines in swap and bankfixed.

jcw, 2019-04-18:
Ah, wait - I'm still getting bank 7 for loading init, even though MAX_MAPS is 5 in my current tests. Odd, `ramsize` is now set to 364, and `procmem` is 300. I've tried `make clean`, just to be sure. So as you pointed out, the swap bank choices seem to be mixed up. I need to figure out where "7" comes from.

EtchedPixels, 2019-04-18:
pagemap_add in discard.c. Are you still adding too many pages  ?

jcw, 2019-04-18:
I've narrowed it down to `ptab_alloc` - added some debug at the end:

```
			udata.u_error = 0;
    kprintf("ptab 1 %x # %d page %d\n", p, p - ptab, p->p_page);
    kprintf("ptab 2 %x # %d page %d\n", newp, newp - ptab, newp->p_page);
	                break;
		}
	}
	irqrestore(irq);
    kprintf("ptab 3 %x # %d page %d\n", newp, newp - ptab, newp->p_page);
	return newp;
}
```

The output I see is this, and that ends up putting 7 in `udata.u_page`:

```
Copyright (c) 2014-2019 Alan Cox <alan@etchedpixels.co.uk>
Devboot
init 1 page 0
ptab 1 B5B3 # 15 page 0
ptab 2 B16C # 0 page 7
ptab 3 B16C # 0 page 7
create 1 page 7
```

Still reading, but if the above rings a bell ...

jcw, 2019-04-18:
Ah, silly me - that sure ain't right:

```
void pagemap_init(void)
{
 int i;
 for (i = 1; i < 8; i++)
  pagemap_add(i);
}
```

jcw, 2019-04-18:
Ok, so that was hard-coded in `platform-z80pack/main.c` - I suggest changing that code to:

```
void pagemap_init(void)
{
    int i;
    for (i = 1; i <= MAX_MAPS; i++)
        pagemap_add(i);
}
```

And now all is well, even with just two user banks:

```
Welcome to FUZIX.
# free
         total         used         free
Mem:       120          120            0
Swap:     3840           60         3780
#
```

I'm closing this issue again. Thanks for the help!

closed 2019-04-18 by jcw