EtchedPixels/FUZIX #703: Q about emulated block device choices

jcw, 2019-04-28:
I'd like to support more disk options in the `jeeretro` build. Since it's based on emulation, there are many ways to do this. Right now, these are the candidates for disk storage:

* stored in internal flash (with a wear-levelling r/w layer) - on ARM and ESP
* stored on SD card, as one-file-per-disk in a FAT filesystem, i.e. disk images
* on ESP, there's also "SPIFFS", which is a file system in flash, again w/ wear-levelling
* perhaps some day: DriveWire, i.e. a fast serial connection
* and some other day: remote drive over Wifi, on ESP32

Some more considerations:

* on ARM, internal flash is very limited on low-end boards, ≥ 1 MB might be usable for FUZIX
* on ESP, there's at least a few MB on SPI flash, but the SPIFFS write access speed is _glacial_
* I now also have my own raw disk wear-levelling layer on ESP flash, and it's sort of doable
* an important aspect of the choices to make will be how easy it is to transfer files in & out
* I'd like to support an 8" SSSD floppy, i.e. 250K, 77 tracks, 26 sec/trk, 128 b/sec, w/ skewing
* for the rest, I'd be fine with all 512 b/sec, LBA,  perhaps just an 1.44 MB floppy + 8 MB HDs
* and then there's the choice of where to put swap, especially with a small number of banks

On the low end, I'm considering ditching support for ARM altogether (e.g. STM32F406VE), and just focusing on ESP32's with either 4 MB flash and SD card, or simply 16 MB flash. Newer ESP boards include 4 or 8 MB RAM, which would be very nice for a large number of memory banks.

My thinking so far is to work with the default `/dev/*` drive definitions in FUZIX, i.e. 4 FD's, and 4 HD's, where the HD's could go all the way up to 15x 8 MB partitions per disk, for SD cards. E.g.

* `fd0` - special-cased 8" SSSD floppy image, with de-blocking "somewhere"
* `fd[1-3]` - 1.44 MB floppy, perhaps one on each medium: raw flash, SPIFFS, SD card
* `hd[0..3]` - always either 8 MB total or ≈ 120 MB total (when partitioned)

Internal flash is too small for a HD image, but SPIFFS is ok (1x 8 MB on an ESP w/ 16 MB flash). Anything larger, i.e. HD's with multiple partitions, will probably only make sense on an SD card (and in the future: DriveWire and/or remotely networked).

So my question boils down to: does anyone have suggestions or comments? I'm not going to implement it all at once, but it would be nice to have a forward-looking growth strategy ...

EtchedPixels, 2019-04-28:
Internal wear levelled flash ought to be fine for CP/M. For Fuzix we'd need to implement 'noatime' otherwise every time you walk a directory or touch a file it will try and update the last accessed timestamps.

Fuzix doesn't really care - give it a raw device with partitions, give it virtual disks, give it a byte stream. Providing it can read and update chunks to give it a 512 byte logical block size it doesn't mind.

CP/M 2.x in particular is happier with 128 byte block sizes but that's a bit hard to do on many media types.

Size limits are 8MB for CP/M 2.x 512MB for CP/M 3 (but good luck using that without directories!) and 32MB for Fuzix currently.

If you have 4-8MB of RAM you might as well make a lot of that ramdisk and maybe battery back it.  There isn't an effective way for a serial based Z80 system to make use of that much space. MP/M goes up to 2MB but that's a 16 user system and that was more theoretical than practical because a single Z80 cannot keep up with 16 users in most environments. When you also have swap it's even more extreme. For 1MB to be useful you need 16 processes actively running at a time which would be slow as molasses.

jcw, 2019-04-28:
Ah, hadn't thought about the timestamp updates - that explains the writes I was seeing. It does make `noatime` support a useful todo item, methinks (nudge). I don't care about access times (but CP/M 2.2's total lack of timestamps makes it very unpractical for dealing with a non-trivial amount of data). That's why to me FUZIX is more convenient for half-serious ~~projects~~ doodling, especially since it can run CP/M 2.2 apps as a task. The CP/M 3 and MP/M variants interest me less than something that feels UNIX'y.

At the moment, I have some builds which launch into FUZIX and different ones for CP/M (both 2.2 and 3.0). I'd like to unify this, and create one setup which starts up as CP/M 2.2, and then launching into the other systems by running the corresponding .COM file. All the basics for this are in place, except for disk mappings and that awkward 128/512-byte difference between basic CP/M and the rest. Adding blocking support to the CP/M resident BIOS is probably the cleanest way to go forward, i.e. keep the emulator 100% 512-byte blocked.

Performance-wise, I don't expect too much from it all, of course. There's also the eZ80 chip, which can run at 50 MHz (and is said to perform as > 100 MHz), and the `socz80` FPGA version. But that's ok, this retro stuff isn't about top CPU performance. It would be nice to have some benchmark baselines, and see how the different ARM & ESP emulations compare (might also be interesting for other CPUs). Some STM32's can go up to 400 MHz and ESP32's are dual-core 240 MHz - this _ought_ to have a decent performance, even via emulation, compared to the old days, when real Z80's were in use.

EtchedPixels, 2019-04-28:
Added an issue for noatime. Some day maybe.

I don't tend to worry about emulator performance too much. If it can emulate a 4MHz Z80 then it's already 'fast' for most of the CP/M period - Z180 and 6MHz Z80B stuff was pretty late - and a lot of the RAM couldn't cope with it anyway at that point. There are plenty of Fuzix optimizations left to do as well.

And if speed was essential why run the emulator layer in the first place 8)

jcw, 2019-04-28:
Exactly - and thanks for #704 anyway.

Right now, my µSD cards appear to be fastest, I've got it to ≈ 1ms read / 2ms write per 512b block. That's pretty snappy, with root and swap mapped as disk images on FAT16. This ESP32 board also happens to have the 4 MB RAM, so a few more banks or swap to RAM will probably be noticeable.

I think I'll drop the internal flash in SPIFFS mode, unless there is no SD card. IOW, use SPIFFS only as fallback when an SD card mount fails, for whatever reason. Both use disk images stored as files.

Any spare RAM can then be used to extend from 3 to say 8 banks, and the rest provided as RAM disk. It shoudln't be too hard to make the kernel adjust its banks to whatever the emulator reports as limit.

And the CP/M BIOS will have to be extended to do 512-byte sector blocking. Then I can merge them.

Thanks for the comments.

jcw, 2019-04-29:
Here's my current / tentative / pie-in-the-sky plan, FWIW:

* fd0..fd2, and hda..hdc are each mapped to files, first searched on SD card, then in SPIFFS
* fd0 = 8" SSSD floppy, somewhat funky access, since 512-byte blocks will span tracks
* fd1 = 1.44 MB floppy, 18 sectors, 160 tracks, probably using the P112 format for CP/M
* fd2 = "any-size" floppy, created empty on startup, expands as it gets written to
* fd3 = spare, perhaps a real floppy drive one day ...
* hda, hdb, hdc = up to 64 MB file, mapped to 8x 8 MB partitions (no MBR, fixed offsets)
* hdd = mapped to spare area on ESP flash, if available (will be > 8 + 1 MB with 16 MB flash)
* rd0 = 1 MB ram disk, mapped to PSRAM when available
* rd1 = 2nd ram disk, mapped to as much PSRAM as remains (2.5 to 6.5 MB)
* all emulator disk accesses are in 512-byte blocks (CP/M will have to block/deblock in its BIOS)

Swap will be configured as `rd0` if it exists (i.e. in ram), or to `hdd2` (in raw flash), or else to `fd2` (dog slow if on SPIFFS), and will be able to hold up to 16 banks, i.e. max somewhat under 1 MB. I understand that the root device can be specified on the CP/M command line when starting FUZIX, but it looks like the swap device is fixed at build time right now. Since with only 2 user banks, swap is needed during boot, maybe I can hack things so a second arg can override the default swap choice. I'll look into it later.

PSRAM, if available, will also be used to expand emulated memory from 3 to 8 banks of 60K, if present.

This should allow running the jeeretro port on a wide variety of ESP32 boards, even from the minimal "4 MB flash w/o SD" to more extended units with a µSD card slot and/or 16 MB flash and/or PSRAM.

For a FUZIX build which can run on any ESP32 board, two disk image files need to be loaded in SPIFFS:

1) a file called `fd0.img` with at least a bootable CPM 2.2 system plus fuzix.com (250 KB)
2) a file called `fd1.img`, containing a small fuzix root fs (1.44 MB)

That, and the z80emu-based app should be all that's needed to get a little FUZIX going on an ESP32.

EtchedPixels, 2019-04-29:
Right now all the SD card using ports can use the same SD card image except for writing boot blocks and kernel to it so it would be good if 'raw sd card' also have a drive. I guess it depends how much you are emulating and how much you are exposing hardware.

If you look at the other platforms quite a few of them support using a swap partition found by scanning the partition table at boot time. That's how most ports actually work. If there is a suitable partition it is used, if not then you either have no swap or die horribly so I guess the obvious way to do it on fuzix is to partition one of the virtual disks.

It's also fairly easy to favour ramdisk. The current ports start with swap_dev = 0xFFFF and when they see a real one use that, but it would thus be trivial to just set swap_dev to the RAM drive if present and 0xFFFF if not and then scan the sd.

Not sure I would swap on raw flash, the swap rate can be quite high if you have few banks.

I can't help feeling any scheme is overly complex.

jcw, 2019-04-29:
My bad - to clarify: "raw flash" is SPI flash with basic wear-levelling on top (an alternative to SPIFFS).  
Ah, yes, with an MBR a partition scan makes sense. Hm, ok, I'll look into that approach as well.

The four ESP32 board/feature combinations i'm currently aiming for are:

* minimal: only 4 MB flash on-board, ≈ 3 MB can be used as file storage via SPIFFS (glacial swaps)
* same, but with 16 MB flash on-board - in this case 12 MB can be used as non-SPIFFS "raw flash"
* with a µSD card socket: that's perhaps the most practical and convenient way to run FUZIX
* with ≥ 4 MB PSRAM (the WROVER modules), some boards also include a µSD card socket

Without PSRAM, I can only fit 3 banks total, 60K each, plus 4K common.

As for ARM, I think STM32F407VE w/ µSD card could still be supported alongside all of the above.

Yes, it's easy to over-engineer it all. But as long as I can make _some_ progress, I'm ok with it.

jcw, 2019-04-30:
Simpler (for me, to implement) would be to always map swap to `/dev/rd1`, and then have the emulator decide which of the available block devices and areas to actually use for swapping. The MBR scheme isn't that useful on SD cards, IMO - swap can just as easily be mapped to an open file on the card.

Also, I'm going to internally renumber all block devices to a single-byte code: 2 bits for fd/hd/ramdisk, 2 bits for subdevice, and 4 bits for (fixed-size) partitions. Along the lines of `dev/z80pack/devfd.c`. That way, the current single disk request to the emulator can continue to be used for all I/O variants.

EtchedPixels, 2019-05-01:
Take a look at dev/blkdev.c.

Z80pack is an old port and a little odd in being a fake platform. Most real world ones use blkdev along with the sd / ide / etc drivers so that it does most of the work for you, manages partitions, scans for swap etc.

nickd4, 2019-05-08:
Hmm I have some ESP32 as well (we use it at work for IoT products), I was thinking more along the lines of repurposing the 680x0 port (has a large linear RAM space in which processes are loaded and relocated to wherever they happen to be loaded, there is no MMU or swap, a process stays where it is loaded until it dies) to run natively on ESP32. I think you might be putting a disproportionate amount of effort into making an elaborate emulator in a rather hostile environment, when it would be so much easier to just run a normal emulator on a Raspberry Pi, for example. Although having said that, the same criticism can be levelled at anyone using ancient hardware so perhaps it is not my place to decide for others where the boundary of "realism" lies. Anyhow I **would** be thrilled to see an ESP32 and/or ARM NoMMU FUZIX port.

jcw, 2019-05-08:
Yep - "because I can" is usually the only meaningful reply here. In my case it's probably a lost youth spent on not being able to afford CP/M. So now, I tend to re-implement Z80 stuff for the heck of it ...

What I like about running on ARM/ESP iso RasPi's etc, is the quick power-up. What I'd _like_ to explore at some point is going for ultra-low power Z80 emulation, but in use and while idling.

The Z80 port is running fine with a bunch of different ESP32 configs. It's going to take more time and brainwaves to figure out how to make this easy to replicate. Most recent code is [here](https://git.jeelabs.org/jcw/retro/src/branch/master/z80emu/cpm3-esp), but no docs and still very much in flux. Getting telnet into the mix would be nice - haven't looked into that yet.

jcw, 2019-05-08:
Closing this issue, as the essence has been resolved now. For the time being, I decided to implement four storage variants: sd (fat), raw spi flash (with my own wear-levelling), spiffs (esp32's variant), and psram. Figuring out how to expose this in the most sensible way is another exercise - for another day.

closed 2019-05-08 by jcw