EtchedPixels/FUZIX #704: Support noatime for flash based storage

EtchedPixels, 2019-04-28:
Provide a way to set a file system to 'noatime', set an inode c_flags for it and catch it in setftime.

beretta42, 2019-04-29:
I was just thinking about this.  Wondering if turning off atime gunk could speed the fs upa bit?

EtchedPixels, 2019-04-30:
Most of our disk devices are ridiculously fast for the CPU (so memcpy speed). Only setting the flag if the time had actually changed would certainly be needed either way. Noatime and variants are just varieties of that. noatime is 'changed by an infinite amount' a 'lazy atime' is changed by a few seconds, to spec is changed by 1 second.

I do need to take a look at updating of atimes when they have not changed. I don't think we cover that and whilst on 1MHz 8bit it's impact isn't big, on 68K or fast 8bit might be.

EtchedPixels, 2021-06-22:
Added

closed 2021-06-22 by EtchedPixels