EtchedPixels/FUZIX #714: what if - from the ground up

jcw, 2019-05-23:
This is completely off-topic in a way, but perhaps an interesting thought experiment nevertheless (if not, please feel free to immediately close this issue, fine with me):

> What would an ideal low-end CPU architecture look like for something Posix'y such as FUZIX?

What I mean is: suppose one were to aim for a small OS such as Fuzix to run on new hardware, designed _specifically_ to make it work optimally. What would it look like? Similar to one of the existing 8-bitters, i.e. 6502/6809/Z80? Would it be a matter of just picking whatever is easiest for a C compiler to generate code for? Are the MMU design choices essential? Would it lead to something similar to (very low-end'ish) ARM or RISC-V? Or be like Oberon?

Evidently, "ideal" is a fuzzy goal to aim for. It could be fast or small code, or easy to emulate, or low gate counts (if done with an FPGA), or easy to grasp by one person, etc.

The reason to ask is that so far, most operating systems appear to have been designed to run on existing hardware (that's what gets real work done, after all). When looking back, and with all the experience collected to date, maybe some good points can be made about what has worked well, but perhaps also what might substantially have affected all the trade-offs.

With the wide scope of architectures covered by FUZIX, and its need to keep things simple due to the limitations of low-end hardware, I think it could be interesting to explore this "what if".

feilipu, 2019-05-23:
I’ve often thought ATmega instruction set on unified memory (not micro controller) would be my choice. Modern 8-bit RISC.

2c.

nick-lifx, 2019-05-25:
It would end up like the PDP-11. Indeed I am told that MSP430 is PDP-11ish, though I have not checked this. The reason is the symmetric instruction set (any operation on any pair of registers), the useful addressing modes corresponding to common C pointer and array ops, and the emphasis on word sized operations (int and pointer are the most common type, much more than byte).

jcw, 2019-06-06:
I'll close this issue. Thx for the comments - we all had our 2 cents, I think we can leave it at that.

closed 2019-06-06 by jcw