EtchedPixels/FUZIX #724: Undocumented instructions in from sdcc

feilipu, 2019-06-25:
I'm compiling using clean repo using `z80pack` target, and I find that compiling `ceilf.c` is causing undocumented assembly using the `iyh` register.

When I do the same with `ceilf.c` with sdcc standalone. i.e with following command line, it doesn't generate the same issue. That, I find really strange.
```sh
sdcc -mz80 --fverbose-asm -S ceilf.c -o ceilf.asm
```
Both results attached.
[`~FUZIX-Library-libs-ceilf.asm`](https://github.com/EtchedPixels/FUZIX/files/3325235/FUZIX-Library-libs-ceilf.asm.txt)
[`~ceilf.asm`](https://github.com/EtchedPixels/FUZIX/files/3325234/test-ceilf.asm.txt)

Normally, this wouldn't be a problem, but I'm trying to compile this for a z180 so this isn't going to work at all. For reference changing the `USERCPU` references to `CPU` in the make files causes the compiler to issue `-mz180`, but this _still_ doesn't prevent undocumented instructions appearing.

There's something afoot, because I didn't just find this bad assembly, it causes an error and causes the compile to bomb too.

```sh
atanhf.c:29: warning 122: dividing by 0
atanhf.c:31: warning 122: dividing by 0
fcc -mz80 -O2 -c -DBUILD_LIBM cbrtf.c -o cbrtf.rel
fcc -mz80 -O2 -c -DBUILD_LIBM ceilf.c -o ceilf.rel
ceilf.asm:193: Error: <a> machine specific addressing or addressing mode error
ceilf.asm:202: Error: <a> machine specific addressing or addressing mode error
ceilf.asm:312: Error: <a> machine specific addressing or addressing mode error
removing ceilf.rel
Makefile.z80:191: recipe for target 'ceilf.rel' failed
make[1]: *** [ceilf.rel] Error 1
make[1]: Leaving directory '/home/phillip/Z80/FUZIX/Library/libs'
make[1]: Entering directory '/home/phillip/Z80/FUZIX/Library/libs'
cp crt0.rel crt0nostdio.rel c.lib /opt/fcc/lib/
cp curses.lib termcap.lib /opt/fcc/lib/
cp m.lib tinymalloc.lib /opt/fcc/lib/
cp: cannot stat 'm.lib': No such file or directory
cp: cannot stat 'tinymalloc.lib': No such file or directory
Makefile.z80:212: recipe for target 'install' failed
make[1]: *** [install] Error 1
make[1]: Leaving directory '/home/phillip/Z80/FUZIX/Library/libs'
Makefile:81: recipe for target 'libs' failed
make: *** [libs] Error 2
phillip@SilverQuad:~/Z80/FUZIX$ 
```
Thoughts? What am I doing wrong?

EtchedPixels, 2019-06-25:
fcc just runs sdcc with options so if this is happening I would suspect sdcc.

If you run fcc with -V as well it will give you the SDCC commands it used. Could well be it's only triggering due to optimization settings and your exact compiler version.

feilipu, 2019-06-25:
The issue seems to be sdcc generating a bad `.rel`, because sdasz80 picks up the problem.

If I compile with the options picked out of `-V` from the make file

- with `-S -o ceilf.asm`, there is no bad code in the resulting `.asm` file,
- but `-c -o ceilf.rel`, it blows up when generating the `.asm`.

See below shell output.

```sh

phillip@SilverQuad:~/Z80/src/test_ceilf$ sdcc -V --std-c99 -mz80 --less-pedantic --max-allocs-per-node100000 --opt-code-size -D__FUZIX__ -Ddouble=float --fverbose-asm -DBUILD_LIBM -I/opt/fcc/include/ -c ceilf.c -o ceilf.rel
+ /usr/local/bin/sdcpp -nostdinc -Wall -std=c99 -D__FUZIX__ -Ddouble=float -DBUILD_LIBM -I/opt/fcc/include/ -obj-ext=.rel -D__SDCC_STACK_AUTO -D__SDCC_CHAR_UNSIGNED -D__SDCC_INT_LONG_REENT -D__SDCC_FLOAT_REENT -D__SDCC=3_9_1 -D__SDCC_VERSION_MAJOR=3 -D__SDCC_VERSION_MINOR=9 -D__SDCC_VERSION_PATCH=1 -D__SDCC_REVISION=11295 -D__SDCC_z80 -D__STDC_NO_COMPLEX__=1 -D__STDC_NO_THREADS__=1 -D__STDC_NO_ATOMICS__=1 -D__STDC_NO_VLA__=1 -D__STDC_ISO_10646__=201409L -D__STDC_UTF_16__=1 -D__STDC_UTF_32__=1 -isystem /usr/local/bin/../share/sdcc/include/z80 -isystem /usr/local/share/sdcc/include/z80 -isystem /usr/local/bin/../share/sdcc/include -isystem /usr/local/share/sdcc/include  ceilf.c 
+ /usr/local/bin/sdasz80 -plosgffw ceilf.rel ceilf.asm
ceilf.asm:193: Error: <a> machine specific addressing or addressing mode error
ceilf.asm:202: Error: <a> machine specific addressing or addressing mode error
ceilf.asm:312: Error: <a> machine specific addressing or addressing mode error
removing ceilf.rel
+ /usr/local/bin/sdasz80 -plosgffw ceilf.rel ceilf.asm returned errorcode 512
phillip@SilverQuad:~/Z80/src/test_ceilf$ 
```

feilipu, 2019-06-28:
As a workaround, `--reserve-regs-iy` is successful for the test file, but I can't find the appropriate place to put the option. Either `fcc` blows up not recognising the option, or `CROSS_CCOPTS` doesn't get applied at the right time to rescue `ceilf.c`.
I'm not sure where is the right place to add the option (without breaking things).

feilipu, 2019-06-30:
Fixed with sdcc [r11311](https://sourceforge.net/p/sdcc/code/11311/).

closed 2019-06-30 by feilipu