EtchedPixels/FUZIX #726: os9 simulator for 6809 fuzix

beretta42, 2019-06-27:
I think it should be possible to run OS9 programs under FUZIX.  The big hurdle would be to simulate everything going on behind OS9's SWI2 interrupt/system call.  Most of the basic IO calls seem to map near directly to unix calls, so that part seem pretty straight forward.  the OS9 function calls are a bit trickier, but trusted OS9 people seem to think only a few of those functions would need some more substantial glue layer.  I do have some questions on how much noodling I can do with the SWI2 interrupt vector.   Some kernel support will be need to switch in/out the each userspace's vector, and possibly for SWI3 too.  Any ideas on allocation of these vectors to userspace?  or do I call this a cheap hack?

EtchedPixels, 2019-06-27:
The Z80 CP/M emulation just saves the kernel syscall vector and uses that to make syscalls because that space is banked. If it was a 68000 or something then yes it ought to catch exceptions and analyse the attempted trap.

For 6809 you might want to have a hook for it in kernel but if your kernel vector jumps to an address that is in the non user owned bits of the writeable banked space (even if the default is just to returns) then you don' t have to do any extra scheduler work just provide an ioctl or something for OS9 emulation.

I've been pondering something similar for some other emulation bits. Perhaps we should have an option to hook devsys,255 as /dev/platform or somesuch ?

Do we need to run Uniflex binaries as well ;-) ?

beretta42, 2019-06-30:
err... yeah.  Whilest perusing OS9 code, I realized I could just have the kernel's swi2 & 3 handlers revector to the current userspace.  Luckily, in coco3, I reserved the low 0x100 bytes in userspace to mirror the Z80 ports' CP/M emulator reservation.  Not sure about the other 6809 platforms though.  The userspace vector being in different spots from platform to platform may confuse the os9 simulator...

Don't forget about plain 'Flex' too! :)

meanwhile here is a quick hack of a os9 simulator.  It's loading os9 brainfuck interpreter, which loads up a bf script:
[image]

EtchedPixels, 2019-07-01:
Cool.

The vector setting wants to go via the kernel precisely because it's non portable (and also so os9sim can just error nicely on other boxes). I will add a platform sysdevice when I get a moment

beretta42, 2019-07-07:
Hey! look who rared his head again: the goofy sbrk()/perror bug from vile debug. ah.. looks like our cpu's brk_limit() is smart. its the current SP minus some space.  so if we move the stack around in user mode, then sbrk can fail.
```#define brk_limit() (udata.u_syscall_sp - 512)```
should be udata.u_utop instead?
```# os9sim cavewalk
h  size: 1b35
h  exec: c61
h dsize: 40b9
ibot: 32e2
itop: 4f00
dtop: 9000
psize: 1
pbot: 8fff
sbot: 8ff3
size: 5d1e
ipc: 3f43
I$Open: "/TERM"
open: 31: out of memory by 25757
Unknown error 2
I$Read: # ```
https://github.com/EtchedPixels/FUZIX/issues/732

EtchedPixels, 2019-07-07:
The current code is right for the normal case. If you are going to move the stack around in the OS9 case then you either want to grab all the memory and manage it yourself, or you need to define a private brk_limit() that honours either sp=512 or in OS/9 mode udata.u_top and knows what mode you are in.
We will also need to be careful to catch that if we optimize some of the swap logic to swap base->brk and sp->top only (ditto fork). I don't think 6809 does at the moment but that wants checking.

beretta42, 2019-07-07:
Thanks. My mind was unclear on a unix kernel's policy on the user SP.   If brk() is the only syscall that worries about where the SP is at, I say leave brk_limit() alone.  a mode-aware brk() sounds overly complex, when it's not hard (and faster) to do brk()'s action in userspace.  I'll just pull the trick you taught me in vile and do one brk() based on a autovariable.

EtchedPixels, 2019-07-07:
Classic Unix believes that the world is

DATA
BSS
Break grows upwards
Gap in address space that even if some of it exists may not behave sanely
RedZone
Stack Pointer
Stack
Top

Red Zone is a weirdness of some systems where you need a bit of space the wrong side of the stack pointer for speed or for some bits of CPU behaviour which is safe to use. In our case right now no CPU has a red zone.

closed 2019-09-15 by beretta42