EtchedPixels/FUZIX #729: vile fails on "big" memory

beretta42, 2019-07-01:
https://github.com/EtchedPixels/FUZIX/blob/25e5fa932484b1638e08b2f9c54ce96b5b935bdb/Applications/util/vile.c#L1531

this brk() fails unless I back &mem down by some 1130 bytes.  Not sure I grok why.

And the following read() fails, as my alloc buffer is more than 32K (and it appears negative to read syscall)

EtchedPixels, 2019-07-01:
The brk failure is weird, can you trace whether the syscall fails and if so why ?

The read one .. yep I need to fix that

beretta42, 2019-07-01:
yes, i can do that.  It took me a minute to figure out why &mem was unvolved! Then i remembered its on the stack way up north.  Lemme do some more checking.

beretta42, 2019-07-02:
I'm smelling a compiler bug.  single step time!

EtchedPixels, 2019-07-02:
Could be.. but could also be the compiler being overly and validly smart

Does it work if you use &argc ?

beretta42, 2019-07-05:
ahah.  found it.  that's not the brk() that's failing.  The doread() now always fails, and does a perror() in the process. perror does a sbrk() to load in the error strings, and that brk() fails. 768 isn't enough to load it.  So brk() works fine, still spent an hour or two single stepping :(  meanwhile doread() doesn't work, but I can submit a patch.

EtchedPixels, 2019-07-05:
Wonder if it's worth teaching strerror to just load the offsets because it's quite a big chunk of space

closed 2019-07-05 by EtchedPixels