EtchedPixels/FUZIX #747: Adventures look to be off by one on screen height

EtchedPixels, 2019-09-23:


tormodvolden, 2019-09-23:
Doesn't look to bad here on dragon-mooh, at least when it comes to the prompt. There is an extra newline printed after my entries but I don't know if that is intended.
[image]
[image]
The "<@>" line is meant to be a separator between upper and lower "window"? I think something like "===" would be nicer.

Would be nice to have some easy-to-run test cases, for verifying vt52 emulation, termcap, etc. I see some oddities in vile and levee but I don't know if it is the application or display layers.

EtchedPixels, 2019-09-23:
The blank line is intended. I am wondering if Brett had the screen height off by one

beretta42, 2019-09-24:
Hmm.. I'll check.  I sorta remember there's some window size api that i never hooked up on the coco3.

tormodvolden, 2019-09-24:
BTW, for the issues I see in vile, I see the same when I compile it for Linux, so it is the application.

EtchedPixels, 2019-10-27:
I think vile is a different set of issues so I am closing this now it seems to be working

There *are* a bunch of other bugs related to word wrapping etc that I do need to tackle and am working on sorting out. I may write myself a split window adventure game library and just stick it all in one place - we shall see

closed 2019-10-27 by EtchedPixels