EtchedPixels/FUZIX #748: vi thinks it successfully saved when it didn't

jes, 2019-10-25:
See save() at https://github.com/EtchedPixels/FUZIX/blob/master/Applications/util/vile.c#L1206 and save_done() immediately after.

In the event that open() returns an error, save() returns -1. save_done() just checks whether save() returned 0, so it thinks -1 is success.

In particular, this means if you open vile, write some text, then ":wq", it just exits with no warning and no error and without saving your changes.

I think most likely save() should initialise ok to 0, but I haven't thoroughly checked this.

jes, 2019-10-25:
It's not as simple as initialising ok to 0 because some of the other logic was wrong as well.

I think this should do it: https://github.com/jes/FUZIX/commit/ef3197786510d8978a12700dfc6ad6e4db2275b2

EtchedPixels, 2019-10-27:
I think I fixed it - rather than invert the logic I used an err instead of ok variable.

closed 2019-10-27 by EtchedPixels