EtchedPixels/FUZIX #752: If you get the username or password wrong, you don't get another try

jes, 2019-10-27:
```
login: roto
Password:

Login incorrect
```

and then it just sits there, and you have to reset the machine. It should give another login prompt.

EtchedPixels, 2019-10-27:
I can't duplicate this - what platform are you trying it on and do you have a working clock (does the time change ?).

jes, 2019-10-27:
It's an RC2014, and no it doesn't have a clock.

I was just looking at the init code and it pause()s after an incorrect login attempt, so that's probably my problem.

EtchedPixels, 2019-10-27:
Yes - if you have no time source then it will hang at the pause as you see so Fuzix needs a clock source card of some kind. Some of the ports also support just wiring an 8 or 10Hz clock to one of the serial port pins. That's not in RC2014 but wouldn't be hard to add as a boot option.

closed 2019-10-27 by EtchedPixels