EtchedPixels/FUZIX #755: loader.ld for platform-atarist?

codesmythe, 2019-11-01:
Hello. I wanted to play around with the Atari ST build of FUZIX, aware of its dev state. 

In the platform-atarist Makefile, the 'loader' target depends on loader.ld, but this file is not in the platform-atarist directory. Perhaps this file did not get committed at some point? Do you have this file available?

Thanks,
Rob

EtchedPixels, 2019-11-01:
Fixed...

it's very dev state. I spent some time working on putting disk and DMA drivers together to emulator level but haven't yet connected them up to the kernel.

closed 2019-11-01 by EtchedPixels

codesmythe, 2019-11-02:
Thanks for the fix. Would you happen to have fuzix.ld for atartist laying around? It's a dependency of the 'image' target.

Thanks again.

EtchedPixels, 2019-11-02:
Doh...