EtchedPixels/FUZIX #756: Installation on ZX Spectrum 128k with DivIDE

BruXy, 2019-11-03:
Hi guys,

What is the best way to try FUZIX on ZXS+128k with DivIDE (version 5.7c with one CF card)? I am currently using ESXDOS 0.8.7.

Regards,
BruXy

EtchedPixels, 2019-11-03:
You need a PC partition table on the CF card

Set up one partition for ESXDOS as normal
Set up one of up to 32MB type 7E and put the Fuzix root file system in it
Set up one of say 4MB type 7F and it will get used for swap

Add the FUZIX command into the ESXDOS commands directory
Add FUZIX.BIN to the top level directory of the ESXDOS partition

Boot into ESXDOS, type '.fuzix', cross fingers, hit enter

It's not an ideal machine to run Fuzix on because of the way the memory banking is in one 16K window, but it should work ok.

BruXy, 2019-11-03:
Hi Alan, thank you for the answer! Not yet successful with build of FUZIX and FUZIX.BIN. I will document some progress here.

# Partitioning

```bash
$ fdisk /dev/sdc

Disk /dev/sdc: 3.7 GiB, 4000317440 bytes, 7813120 sectors
Disk model: Compact Flash   
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device     Boot   Start     End Sectors  Size Id Type
/dev/sdc1          2048  264191  262144  128M  b W95 FAT32
/dev/sdc2        264192 6555647 6291456    3G  b W95 FAT32
/dev/sdc3       6555648 6621183   65536   32M 7e unknown
/dev/sdc4       6621184 6629375    8192    4M 7f unknown
```
# Create file system

From: [Z80 Root Filesystem (32MB)](http://www.fuzix.org/downloads/0.3/rootfs-z80-32.gz)

```bash
$ wget http://www.fuzix.org/downloads/0.3/rootfs-z80-32.gz
$ gunzip rootfs-z80-32.gz
$ dd if=rootfs-z80-32 of=/dev/sdc3
 
65535+0 records in
65535+0 records out
33553920 bytes (34 MB, 32 MiB) copied, 9.36928 s, 3.6 MB/s
```
**TODO:** Does 4M swap need some formatting?

# Build on Fedora Linux

Installation of compilers: `dnf install z88dk sdcc byacc`
Fedora's SDCC package provides binaries with sdcc- prefix, create symlinks:

```
ln -s /usr/bin/sdcc-sdasz80 /usr/local/bin/sdasz80
ln -s /usr/bin/sdcc-sdar /usr/local/bin/sdar
git clone https://github.com/EtchedPixels/FUZIX.git
```

Edit `Makefile`: `TARGET=zxdiv`

**TODO:** Main build went until linker died with (`sdcc` is 3.8.0) during linking:

```
make[2]: Entering directory '/mnt/old_root/usr/local/src/FUZIX/Applications/util'
../../Library/tools/fcc -mz80   banner.rel -o banner

?ASlink-Warning-Undefined Global '__divuint' referenced by module 'fread'

?ASlink-Warning-Undefined Global '__mulint' referenced by module 'fread'

?ASlink-Warning-Undefined Global '__modulong' referenced by module 'ltostr'

?ASlink-Warning-Undefined Global '__divulong' referenced by module 'ltostr'
...etc...
```

BruXy, 2019-11-04:
My build is currently failing on this (any idea @EtchedPixels ?):

```
make[2]: Leaving directory '/mnt/old_root/usr/local/src/FUZIX/Kernel/cpm-loader'
cc     tools/trslabel.c   -o tools/trslabel
cp hogs.txt hogs.txt.old
cp: cannot stat 'hogs.txt': No such file or directory
make[1]: [cpu-z80/image.mk:52: fuzix.bin] Error 1 (ignored)
tools/memhogs <fuzix.map |sort -nr >hogs.txt
head -5 hogs.txt
24576: _bufpool
9724: _keymap
5465: _sd_spi_transmit_sector
1500: _i_tab
1271: __execve
tools/bihx fuzix.ihx
tools/binprep
Bad format for relocated long call at c306 (e3 cd f8 d2 cd
make[1]: *** [cpu-z80/image.mk:56: fuzix.bin] Error 1
make[1]: Leaving directory '/mnt/old_root/usr/local/src/FUZIX/Kernel'

```

EtchedPixels, 2019-11-04:
The zxdiv needs the modified sdcc not the Fedora one. The Fedora sdcc doesn't support banked binaries which is how I stuffed it all into a 128K machine with 16K bankable space.

You need the sdcc from my github.

BruXy, 2019-11-06:
Hi @EtchedPixels 

Fingers crossed, FuzixOS is running. I have updated a comment above with the installation steps.

[image]

Thank you for your help. Happy hacking!

BruXy, 2019-11-06:
Compiled FUZIX and FUZIX.BIN for target platform "zxdiv": 

http://bruxy.regnet.cz/8bit/fuzix/

EtchedPixels, 2019-11-06:
Yay...

vi is a bit too big - the 128K spectrum I can only really get a 32K user space. I'm trying to squash vi down a bit more but the first squash, removing curses means there is a lot of debugging left to do!

ed should work fine.

closed 2019-11-25 by BruXy