EtchedPixels/FUZIX #757: v68: can't load /etc/inittab

codesmythe, 2019-11-03:
Backing up from the Atari ST platform, I thought I'd give the v68 platform a try. The first thing I see is that init can't find /etc/inittab:

    robg@ubuntu:~/Virtual68$ ./v68
    V68 Boot Services
    Revision 0.1 Alpha
    
    Clearing memory
    41e: BUS 524288
    41e: BUS 524289
    41e: BUS 524290
    41e: BUS 524291
    424: BUS 524288
    424: BUS 524289
    424: BUS 524290
    424: BUS 524291
    ide0: ACME NEMESIS RIDICULII v0.1
    ide1: not present
    > b
    Booting from IDE disk 0
    OK
    Loading...OK
    FUZIX version 0.4pre1
    Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
    Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
    Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
    Copyright (c) 2014-2019 Alan Cox <alan@etchedpixels.co.uk>
    Devboot
    512kB total RAM, 448kB available to processes (15 processes max)
    Enabling interrupts ... ok.
    IDE reset
    IDE drive 0: hda: hda1 hda2 
    IDE drive 1: ide error, status=0000
    bootdev: 2
    Mounting root fs (root_dev=2, ro): OK
    Starting /init
    init version 0.9.0ac#1
    init: no inittab
    sh: Unknown error 14

    panic: killed init

I made an attempt to see if I could debug this myself. Init sees the failure here [1] when fstat() returns -1. Fstat() returns this because the usermem.c:valaddr() check returns zero: The st stat struct allocated in init.c is given an address like 0x0000BD36, which is less than the PROGBASE (0x20000) defined in platform-v68/config.h. So when  attempting to copy the stat back to user memory, the bounds check in valaddr [2] fails.

That's as far as I was able to go. I don't know why init has chosen that address. Also, the v68 target seems to have been configured for a single process at time(?), as opposed to tiny68k, for example.

[1] https://github.com/EtchedPixels/FUZIX/blob/master/Applications/util/init.c#L405
[2] https://github.com/EtchedPixels/FUZIX/blob/master/Kernel/usermem.c#L21

EtchedPixels, 2019-11-03:
Ah that would be d13570c7f91981b9f60825b813cee40a3dc1b2b8. I've not tested 68000 in a bit as I've been working on Z80 primarily whilst adding the 0.4 stuff

BD36 will be the supervisor stack...

Patch pushed

v68 is configured as a single process, small memory machine. It's very loosely based on the Retrobrew 68000 board hence the ability to set the low memory supervisor only. Tiny68K is a 16MB 68000/68010 system by Bill Shen. It's got an emulator in the v68 directory as well.

I've been using tiny68k for the initial 68000 work because it's simple - serial, timer, CF card and loads of RAM. No complicated or weird peripherals and no particularly awkward interrupt logic.

codesmythe, 2019-11-03:
That fixed it. I can now boot to a shell. That was quick!

The tiny68k emulator is looking for a tiny68.rom file. Do you know where I can get that? Is that from Bill Shen? A renamed boot.dat?

[BTW, my motivation for this 68K stuff that I have my own retrobrew 68K board that I'd like to port FUZIX to eventually (the board can run CP/M-68K now). The peripherals (IDE, MFP, RTC, Z85C30 serial ports) are mapped to same addresses as the Falcon030, which is why I started looking at the ST stuff.  (Was originally going to do an '030 board, but simplified to 68000).]

EtchedPixels, 2019-11-03:
https://www.retrobrewcomputers.org/doku.php?id=boards:sbc:tiny68k

https://www.retrobrewcomputers.org/lib/exe/fetch.php?media=boards:sbc:tiny68k:t68kbug_bin.zip

For the peripherals the MFP and IDE should be fairly easy. RTC always depends what it is and how it's wired. There is Z80SIO code in some of the other ports but no  Z85C30 yet.Tiny68K is a much smaller cleaner tree so probably the better reference.

EtchedPixels, 2019-11-03:
Oh and to save you much grief if you didn't already notice - the IDE on tiny68K is byteswapped

closed 2019-11-12 by EtchedPixels