EtchedPixels/FUZIX #761: z80pack - 0049_switchin: FAIL

Qcraft, 2019-11-28:
I am trying to run fuzix on z80pack

Here is how i created the disk images:

`dd if=bootblock.bin of=drivea.cpm conv=notrunc`
`dd if=fuzix.bin of=drivea.cpm bs=1 seek=193024 conv=notrunc`

`./build-mini-filesystem  drivei.cpm 64 2880`

and here is the output of ./cpmsim:

`FUZIX version 0.4pre1`
`Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza`
`Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha1`
`Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>`
`Copyright (c) 2014-2019 Alan Cox <alan@etchedpixels.co.uk>`
`Devboot`
`484kB total RAM, 420kB available to processes (15 processes max)`
`Enabling interrupts ... ok.`
`bootdev: 0`
`Mounting root fs (root_dev=0, ro): OK`
`Starting /init`
`init version 0.9.0ac#1`
`Checking root file system.`
`0049_switchin: FAIL`

`panic: MMU trap/bad stack`
`Op-code trap at c916 ed 81`

Am I doing something wrong?

EtchedPixels, 2019-12-02:
Probably not - the head of the tree hasn't been tested on z80pack

Qcraft, 2019-12-02:
Ok, so I should just wait for 0.4?

What happened to the 0.3 tag?

EtchedPixels, 2020-03-24:
Fixed

closed 2020-03-24 by EtchedPixels