EtchedPixels/FUZIX #763: msx2 rom compilation stuck on sdcc error

facmachado, 2019-12-10:
Good night to all,

I'm having a problem on compiling the fuzix msx2 rom. I'm getting an sdcc syntax error but the syntax looks correct, and don't know where to go from this point.

Thank you for help. My screenshot goes below:
> Debian stable
> SDCC 3.9.0

```
root@host:/usr/src/FUZIX# cat /etc/debian_version
10.2

root@host:/usr/src/FUZIX# sdcc -v
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ez80_z80/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8/pdk14/pdk15 3.9.0 #11195 (Linux)
published under GNU General Public License (GPL)

root@host:/usr/src/FUZIX# grep TARGET Makefile 
TARGET=msx2
include Kernel/platform-$(TARGET)/target.mk
# TARGET is what we are building
export TARGET CPU USERCPU PATH FUZIX_ROOT

root@host:/usr/src/FUZIX# make kernel
(cd Library; make; make install)
make[1]: Entering directory '/usr/src/FUZIX/Library'
cc -I../Kernel/include -o tools/syscall tools/syscall.c
cc -I../Kernel/include -o tools/binman tools/binman.c
cc -I../Kernel/include    tools/fcc.c   -o tools/fcc
cc -I../Kernel/include -o tools/syscall_6502 tools/syscall_6502.c
cc -I../Kernel/include -o tools/syscall_68000 tools/syscall_68000.c
cc -I../Kernel/include -o tools/syscall_6809 tools/syscall_6809.c
cc -I../Kernel/include -o tools/syscall-scc6809 tools/syscall-scc6809.c
cc -I../Kernel/include -o tools/liberror tools/liberror.c
make[1]: Leaving directory '/usr/src/FUZIX/Library'
make[1]: Entering directory '/usr/src/FUZIX/Library'
mkdir -p /opt/fcc/bin /opt/fcc/lib /opt/fcc/include
mkdir -p /opt/fcc/include/6502 /opt/fcc/include/arpa
mkdir -p /opt/fcc/include/msp430x /opt/fcc/include/netinet
mkdir -p /opt/fcc/include/readline
mkdir -p /opt/fcc/include/sys
install -m 0755 tools/binman tools/relocbin tools/fcc /opt/fcc/bin
install -m 0644 include/*.h /opt/fcc/include
install -m 0644 include/6502/*.h /opt/fcc/include/6502
install -m 0644 include/arpa/*.h /opt/fcc/include/arpa
install -m 0644 include/msp430x/*.h /opt/fcc/include/msp430x
install -m 0644 include/netinet/*.h /opt/fcc/include/netinet
install -m 0644 include/readline/*.h /opt/fcc/include/readline
install -m 0644 include/sys/*.h /opt/fcc/include/sys
install -m 0644 ../Kernel/include/drivewire.h /opt/fcc/include/sys/
install -m 0644 ../Kernel/include/userstructs.h /opt/fcc/include/sys/
make[1]: Leaving directory '/usr/src/FUZIX/Library'
(cd Kernel; make)
make[1]: Entering directory '/usr/src/FUZIX/Kernel'
platform-msx2/image.mk:2: warning: overriding recipe for target 'fuzix.ihx'
cpu-z80/image.mk:49: warning: ignoring old recipe for target 'fuzix.ihx'
platform-msx2/image.mk:6: warning: overriding recipe for target 'fuzix.bin'
cpu-z80/image.mk:52: warning: ignoring old recipe for target 'fuzix.bin'
cc     tools/makeversion.c   -o tools/makeversion
tools/makeversion "0.4pre1" "" msx2
rm -f platform
ln -sf platform-msx2 platform
make -C platform-msx2
make[2]: Entering directory '/usr/src/FUZIX/Kernel/platform-msx2'
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c ../dev/devsd.c
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c ../dev/mbr.c
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c ../dev/blkdev.c
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c devfd.c
devfd.c:27: warning 84: 'auto' variable 'block_xfer' may be used before initialization
devfd.c:27: warning 84: 'auto' variable 'block_xfer' may be used before initialization
devfd.c:32: warning 85: in function fd_transfer unreferenced local variable : 'dptr'
devfd.c:32: warning 85: in function fd_transfer unreferenced local variable : 'dlen'
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c devhd.c
devhd.c:27: warning 84: 'auto' variable 'block_xfer' may be used before initialization
devhd.c:27: warning 84: 'auto' variable 'block_xfer' may be used before initialization
devhd.c:32: warning 85: in function hd_transfer unreferenced local variable : 'dptr'
devhd.c:32: warning 85: in function hd_transfer unreferenced local variable : 'dlen'
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c devlpr.c
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c devices.c
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c main.c
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c devtty.c
devtty.c:116: warning 85: in function tty_data_consumed unreferenced function argument : 'minor'
devtty.c:125: warning 85: in function tty_setup unreferenced function argument : 'flags'
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/usr/src/FUZIX/Kernel/cpu-z80 -I/usr/src/FUZIX/Kernel/platform-msx2 -I/usr/src/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /usr/src/FUZIX/Kernel/cpu-z80/switch.peep  -I../dev/ --codeseg CODE2 -c ../dev/rp5c01.c
/usr/src/FUZIX/Kernel/cpu-z80/cpu.h:6: syntax error: token -> 'irqflags_t' ; column 27
make[2]: *** [Makefile:27: ../dev/rp5c01.rel] Error 1
make[2]: Leaving directory '/usr/src/FUZIX/Kernel/platform-msx2'
make[1]: *** [Makefile:106: target] Error 2
make[1]: Leaving directory '/usr/src/FUZIX/Kernel'
make: *** [Makefile:89: kernel] Error 2

root@host:/usr/src/FUZIX# head Kernel/cpu-z80/cpu.h
#define uputp  uputw			/* Copy user pointer type */
#define ugetp  ugetw			/* between user and kernel */
#define uputi  uputw			/* Copy user int type */
#define ugeti  ugetw			/* between user and kernel */

typedef uint16_t irqflags_t;

extern void out(uint8_t addr, uint8_t val);
extern uint8_t in(uint8_t addr) __z88dk_fastcall;
```

EtchedPixels, 2019-12-16:
Fixed - MSX2 is still somewhat bitrotted though

closed 2019-12-16 by EtchedPixels

facmachado, 2019-12-17:
It worked, thank you! :+1: