EtchedPixels/FUZIX #765: how to contribute

jcw, 2020-01-20:
I'd like to gain a better understanding of FUZIX internals (and have kernel driver experience with unix v7, or at least I did long ago...). So maybe I can help out somewhere. The silly question is: how? What would be a good place to start and what might be a worthwhile issue to tackle?

There's #555 tagged as "help wanted", but for starters, it might be better for me to work on something which does not require setting up a special hardware configuration. I have a diy ez80 board and now also an SC130 Z180-based board.

Then again, perhaps a build running on an emulated rc2014 or z80 might be a much more effective way to move forward, given that this can lead to much snappier edit-run-debug cycles?

EtchedPixels, 2020-01-21:
The SC130 is fairly well covered, the ez80 stuff is a bit minimal right now so fleshing that out to be useful would be one thing.

I do most of my debug on the rc2014 emulation. For one I can collect things like I/O and instruction traces and then go back and see where it went wrong when doing kernel stuff. For user space it's not generally such an issue although as I don't have a ptrace() debugging native is a bit of a pain.

jcw, 2020-01-21:
Ok, I'll see if I can dig a bit deeper into the ez80 a bit more again. Thx.

closed 2020-01-21 by jcw