EtchedPixels/FUZIX #767: Questions about 68HC11 support

jimmoores, 2020-01-25:
I've been following Fuzix for some years (thanks for such a fantastic project!), and wanted to build some hardware to run it on. I've started building a MC68HC11 SBC (based on the ROMless F0 variant), but I need a bit more information than I can find in the source/docs about it's current state and hardware required for the test platform.

- From what I can figure out, the console is through the SCI ports and the SD card is hooked up to the SPI interface?
- The memory map in the README describes a 2K EEPROM at the top of the memory map (F800-FFFF), the I/O mapped to F000-F03F, the internal RAM from 0000-00FF and external RAM from 0100-F7FF.
- Looking through the code, I can see that actually, the external RAM should be banked from 0100-EFFF and unbanked from F040-F0FF.
- Where I get confused is in this section:

```
  page0 (rwx) : ORIGIN = 0x00000, LENGTH = 0x0100
  ram   (rwx) : ORIGIN = 0x00100, LENGTH = 0xEE00
  ports (rw)  : ORIGIN = 0x0F000, LENGTH = 0x0040
  iram  (rwx) : ORIGIN = 0x0F040, LENGTH = 0x00C0
  ram2  (rwx) : ORIGIN = 0x0F100, LENGTH = 0x0500
  eeram (rwx) : ORIGIN = 0x0F600, LENGTH = 0x0200
  eeprom      : ORIGIN = 0x0F800, LENGTH = 0x0800
```

Clearly `iram` is the non-paged RAM block, but what is `ram2` and `eeram`? They seem to be mapped to a data segment in the linker script, but that's about all I can make out.

It'd also be useful to know if anyone is aware of the current working state of the HC11 stuff?  I've seen it marked as 'test' and can see it's not changed much.  Keep in mind that I'm not expecting much, just whether I can initialise something and perhaps contribute back.

Any help would be very much appreciated, apologies if this isn't the best place to ask, but can't find any mention of a mailing list.

EtchedPixels, 2020-01-26:
I sketched it out and built an image to prove the compiler and toolchain actually seemed to work, then started on a 68HC11 board design which has been somewhat in the background (plus version 1 didn't work, version 2 only works internal ram/rom and I'm currently waiting to get more of the house back in order so I can debug version 3). I have a 6303 board working now with RC2014 so I'm hoping I'll also be able to do an RC2014 HC11 pretty much off the same design.

The sketched design is roughly what the board implements but not actually the same in the end. SD card via the SPI interface, console via the SCI, GPIO pins drive the upper bits. I got sufficiently annoyed trying to find a 68HC811 at a sensible price that I added an actual external EPROM to the v3 board which can be paged in/out via an I/O pin.

The linker script is software not hardware areas

page0 is obvious
ram is where the programs and banked part of the kernel go
ports/iram are the internal RAM and I/O (which hide a bit of real RAM)
ram2 is so that there is somewhere to direct code/data that must be common
eeram/eeprom are for the 68HC811 idea

There is a fuzix-devel google groups list although its also usually very quiet.

jimmoores, 2020-01-26:
Hi, many thanks for the prompt reply.  I think that clears up most of my questions.  I'll see how I get on.

closed 2020-01-26 by jimmoores