EtchedPixels/FUZIX #768: Serial port support?

pawosm-arm, 2020-02-08:
I can't find it easily, but it seems like Fuzix for ZX Spectrum +3 lacks serial port support and ability to redirect console to serial port. Such ability exists in CP/M Plus for ZX Spectrum +3, it is capable to make use of built-in serial port and by allowing redirecting console to it, it offers surprising capabilities to this age old system.

EtchedPixels, 2020-02-08:
The spectrum port is a bitbang port without interrupts. That works for CP/M which is polled but doesn't work for receive on a multitasking system. At least I've yet to figure out a way to make it usable.

With an interrupt it's sort of doable but burns a lot of CPU, although I've not actually implemented that for much because it's also a complete pain in the arse 8)

pawosm-arm, 2020-02-08:
That's what I assumed, bit-banging serial port driver would have to work within long periods of disabled interrupts, disrupting perception of time throughout all of the system (yet I wonder how annoying it would be for a system that is not supposed to run too many things simultaneously). For a long time I presumed it couldn't be achieved even on CP/M. Before I discovered that `device` command can redirect console to the +3's built-in serial port, I thought this port is only good for sending text to printers (and the +3 serial port emulation in Fuse can only do that, although the role of a printer plays a single plain-text file) or transferring files between two machines, I was pretty certain that interactive sessions are not possible!

EtchedPixels, 2020-02-09:
The problem is the input side. Output is not a big problem - in fact at high baud rate it's not much different bitbanging to waiting for the uart, and it does support one of Bill Shen's designs that does this.

On input however you have no interrupt on the spectrum to even look for serial I/O. CP/M is OK because apps that want call into the BIOS each time to poll or block on a read.

pawosm-arm, 2020-02-10:
This issue needs some other point of view. How I planned to use it in a first place?:
1. to copy files (or data in general) between pc and fuzix,
2. to have remote access the good old speccy when I'm at work (it's possible now with CP/M: ssh to my linux box from work, then use either `screen` or `minicom` to connect CP/M console redirected to the speccy serial port),
3. to have TCP/IP networking over an ppp interface (if there's a plan to have one in Fuzix).

Seems like it can't be achieved with Speccy's internal bit-banging serial port without sacrificing Fuzix's multithreading. But what about other hardware solutions? I have Spectranet device that is left completely unused. The main reason for it is that its firmware limits capabilities of +2A/+2B/+3/+3B/+3e machines severely. Fortunately, the firmware can be disabled with a jumper, otherwise I wouldn't be able to boot CP/M or Fuzix with this device attached. According to the manufacturer, this jumper does not disable the hardware, it's still I/O-accessible, yet one needs to write their own OS (or a driver to an existing OS) to control this device. It may require a lot of work (hence it remains unused in my Speccy's setup), but I wonder, maybe Fuzix is a good platform for doing this? Having networking interface, a telnet server and an ftp server could be developed for Fuzix. Plus Spectranet offers extra 128kB of 4kB-paged extra memory (as described here: [http://spectrum.alioth.net/doc/index.php/Using_paged_RAM](http://spectrum.alioth.net/doc/index.php/Using_paged_RAM)).

EtchedPixels, 2020-02-10:
I tend to use Standalone/ucp on the PC end to copy files and stuff.

Spectranet is a WizNet 5100 I believe ? I do have some fairly minimal WizNet5100 support compiled in for some platforms that then gets exposed as a socket API. Might be a tight squeeze to fit that into a Spectrum +3. FUSE has some emulation - never used it in Fuse but I re-used their WizNet emulation code in the RC2014 emulator 8)

Added as #769

pawosm-arm, 2020-02-10:
According to SinclairFAQ it is WizNet5100 https://faqwiki.zxnet.co.uk/wiki/Spectranet

EtchedPixels, 2020-02-10:
Looks doable providing the spectranet handles the +3 memory map correctly (ie it pages in over the ROM if the ROM low memory range is triggered, not just if you access 3E00-3FFF in CP/M memory modes)[](url)

pawosm-arm, 2020-02-10:
I guess enforcing correct memory mapping is the purpose of yet another jumper on Spectranet: ZX +2A/+3

This ticket goes into massive off-top and I doubt the original matter will be discussed any longer. Shall this one be closed?

closed 2020-02-10 by EtchedPixels