EtchedPixels/FUZIX #771: 6809 exec changes broke stuff

EtchedPixels, 2020-02-11:
So says @beretta42 and I don't want to lose the fact this needs fixing.

beretta42, 2020-02-13:
Its a moving bug.. the minute i put a printf in sh it errors at a different spot. Yay.  I'm still perusing syscall debug output, but not seeing anything too exciting (yet)  init seems ok, it spawns login and takes 'root' dump motd, etc.

EtchedPixels, 2020-02-13:
That's what I'd expect to see if it's getting the binary loading wrong 8(

More useful might be to make the kernel dump the addresses of the top of code, data, bss, stack pointer for an old working tree and the current one.

EtchedPixels, 2020-03-30:
I've pushed a fix I think for this. The crt0 code didn't include the header in the code size which in turn meant that the bss started overlapping the last 18 bytes of the data segment and things went down the pan from there.

beretta42, 2020-03-30:
:( still dying as before:
[image]

beretta42, 2020-04-01:
Sorry it took a while.  Real life's been distracting.

closed 2020-04-12 by EtchedPixels