EtchedPixels/FUZIX #772: Feature request: extend standalone desktop-side ucp and fsck tools with ability to operate on +3 .dsk images formatted with Fuzix filesystem

pawosm-arm, 2020-02-13:
As in the topic. One can format an empty disk image file in the `Gotek` interface attached to the real Spectrum +3, which operates on the +3 `.dsk` disk image files and then copy that file to the PC. Alternatively, one can patch FUSE emulator with [this patch](https://sourceforge.net/p/fuse-emulator/bugs/_discuss/thread/7c3f1f70fe/f963/attachment/diff.upd_scan_bug.patch) which fixes disk formatting capability of FUSE (still one needs to provide existing `.dsk` image file, e.g. created by one of `libdsk-utils`, as formatting a fresh image started by selecting "Insert New" option still produces faulty `.dsk` files). Such filesystem can then be mount in Fuzix, files can be stored, etc.

Unfortunately, there's no UNIX-desktop-side tools for handling `.dsk` image files holding Fuzix filesystem, similar to standalone `ucp` and `fsck` that currently can operate on a raw Fuzix filesystem only.

EtchedPixels, 2020-02-13:
Turns out to be quite easy so added (needs testing for the libdsk cases)

pawosm-arm, 2020-02-13:
I tried to use newly build `ucp` on some `.dsk` image ([fuzixfs.dsk.gz](https://github.com/EtchedPixels/FUZIX/files/4201717/fuzixfs.dsk.gz) gzipped for GitHub to be happy with it) formatted under Fuzix containing an `file.txt` file. Unfortunately, `ucp` can't handle it:
```
$ ./ucp fuzixfs.dsk 
Opening fuzixfs.dsk (offset 0)
panic: no filesys
```

EtchedPixels, 2020-02-13:
You need to ucp libdsk:fuzixfs.dsk, although it would still have failed

I can now read your test image ('this is a file') with ucp. Some other stuff fails so I don't think we are quite there yet.

pawosm-arm, 2020-02-23:
Seems very close, some things work, some things don't. Yet one thing I can't find in the interactive `ucp` tool: how to save the changes? E.g. I've created a directory and left the `ucp`, opened my `.dsk` image again, and the directory wasn't there...