EtchedPixels/FUZIX #774: Hitting the wall with sdcc 3.8.3 #10676 (Linux)

jayacotton, 2020-03-19:
make[2]: Entering directory '/home/jay/projects/FUZIX/Kernel/platform-rc2014'
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/home/jay/projects/FUZIX/Kernel/cpu-z80 -I/home/jay/projects/FUZIX/Kernel/platform-rc2014 -I/home/jay/projects/FUZIX/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /home/jay/projects/FUZIX/Kernel/cpu-z80/switch.peep  --external-banker -I../dev/ -I../dev/net/ -I../dev/rc2014/ --codeseg CODE1 -c devgpio.c
at 1: warning 117: unknown compiler option '--external-banker' ignored
devgpio.c:103: error 9: FATAL Compiler Internal Error in file 'gen.c' line number '4128' : Register parameter overwrites value that is still needed 
Contact Author with source code

jayacotton, 2020-03-19:
I saw a comment that you had a hack for register passing, is that for this issue ?

EtchedPixels, 2020-03-19:
The internal error is the compiler version you are using
The --external-banker is because you need my patched sdcc that supports banked executables as the RC2014 kernel is banked. See https://github.com/EtchedPixels/sdcc280

closed 2020-03-24 by EtchedPixels