EtchedPixels/FUZIX #775: in platform-amstradnc/config.h no CONFIG_RTC_INTERVAL etc.

tinkercnc, 2020-03-19:
Compiling for the NC100 fails in KERNEL/timer.c due to the lack of  CONFIG_RTC_INTERVAL.
I don't know if this hack below is ok but now it compiles.

```
diff --git a/Kernel/platform-amstradnc/config.h b/Kernel/platform-amstradnc/config.h
index 2dceef5b..97ec3d9e 100644
--- a/Kernel/platform-amstradnc/config.h
+++ b/Kernel/platform-amstradnc/config.h
@@ -3,6 +3,8 @@
 
 /* We have an RTC */
 #define CONFIG_RTC
+#define CONFIG_RTC_INTERVAL 100
+
 #define CONFIG_RTC_FULL
 /* Enable to make ^Z dump the inode table for debug */
 #undef CONFIG_IDUMP
```

And IMHO this was upside down.

```
diff --git a/Kernel/platform-amstradnc/devgfx.c b/Kernel/platform-amstradnc/devgfx.c
index 32694ff3..7f45f31c 100644
--- a/Kernel/platform-amstradnc/devgfx.c
+++ b/Kernel/platform-amstradnc/devgfx.c
@@ -8,29 +8,30 @@
 #include <graphics.h>
 #include <devgfx.h>
 
-#ifdef CONFIG_NC100
+
+#ifdef CONFIG_NC200
 static struct display ncdisplay = {
   0,
-  480, 64,
-  512, 64,
+  480, 128,
+  512, 128,
   0xFF, 0xFF,          /* For now */
   FMT_MONO_WB,
   HW_UNACCEL,
   GFX_TEXT,
   0,
-  GFX_DRAW
+  GFX_DRAW,
 };
 #else
 static struct display ncdisplay = {
   0,
-  480, 128,
-  512, 128,
+  480, 64,
+  512, 64,
   0xFF, 0xFF,          /* For now */
   FMT_MONO_WB,
   HW_UNACCEL,
   GFX_TEXT,
   0,
-  GFX_DRAW,
+  GFX_DRAW
 };
 #endif

```

And booting the resulted images with nc100emu gives:
```
tnc100em -p nc100emu.bin
UZIFUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2020 Alan Cox <alan@etchedpixels.co.uk>
Devboot
256kB total RAM, 192kB available to processes (15 processes max)
Enabling interrupts ... ok.

Baud Rate Generator: 86
IN from unknown Port: &cf
IN from unknown Port: &ce
IN from unknown Port: &cd
IN from unknown Port: &cc
IN from unknown Port: &cb
IN from unknown Port: &ca
IN from unknown Port: &c9
IN from unknown Port: &c8
IN from unknown Port: &c7
IN from unknown Port: &c6
IN from unknown Port: &c5
IN from unknown Port: &c4
IN from unknown Port: &c3
IN from unknown Port: &c2
IN from unknown Port: &bf
IN from unknown Port: &be
...
..
.
```
\>  20000 lines
and than it hangs.
And what has surprised me is the message:

"256kB total RAM, 192kB available to processes (15 processes max)"

because it's definitively " TARGET=amstradnc/nc100" in the Makefile.
But I'm a noob so maybe its ok.

Can someone send me a working PCMCIA image and nc100emu.bin for testing. 

Regards 

tinkercnc aka Matsche

EtchedPixels, 2020-03-19:
I've not tested nc100 for some time so it's probably not well from all the changes and indeed it looks like it goes castors up when it takes an interrupt. The patches look right (RTC interval requirement is a new one from 0.4 changes)

The 256K RAM is right. Fuzix runs entirely off the PCMCIA card as the NC100 and NC200 map the PCMCIA card as memory so it runs with 256K of of the card as memory and 768 as the file system.

tinkercnc, 2020-03-19:
OK, so I suggest to add some tags or branches where the different plattforms are working.
It would much easier to dissect such probs.

But anyway, do you have some suggestions how I can debug the probs above?
Now I've got some sram cards so I can play on bare metal. :)

BTW: Did you get my PM?

EtchedPixels, 2020-03-24:
I'd debug on the emulator first. I did get your message but life around here is a bit more complicated than usual.

I've pushed a fix to one thing broken in nc100.s that I was reminded about by the MSX2 fixes.

tinkercnc, 2020-03-24:
Yes I do.

Same here but I'm in the boondocks and it's much more quiet. ;)

Yesterday I tried this :
```diff --git a/Kernel/platform-amstradnc/tricks.s b/Kernel/platform-amstradnc/tricks.s
index 6e22e9eb..2a5d5c2a 100644
--- a/Kernel/platform-amstradnc/tricks.s
+++ b/Kernel/platform-amstradnc/tricks.s
@@ -26,7 +26,8 @@ fork_copy:
        ld de, #P_TAB__P_PAGE_OFFSET
        add hl, de
        ; hl now points into the child pages
-       ld de, #U_DATA__U_PAGE
+       ;ld de, #U_DATA__U_PAGE
+       ld de, #_udata + U_DATA__U_PAGE
        ; and de is the parent
 fork_next:
        ld a,(hl)
 ```
I did it on the wrong file. Thanks for clarification. :-)

But whats about my Kernel/platform-amstradnc/devgfx.c changes?
IMO it's  distorted. The NC100 has 64 rows and the NC200 128.
Or I'm missing something?

EtchedPixels, 2020-03-24:
You want that one as well..

Not checked on the video bit yet

EtchedPixels, 2020-03-24:
devgfx.c looks right - ifdef NC100 64 rows, else 128 rows ?

(and I've pushed tricks.s with your fix and one more)

tinkercnc, 2020-03-25:
Hmmm, but where is CONFIG_NC100 defined?

Thanks!

EtchedPixels, 2020-03-25:
Yes that got broken in the move to amstradnc/*
Fixed

EtchedPixels, 2021-02-19:
Ok the base issue is fixed so closing this - there is still stuff to fix for 0.4 in the port though.

closed 2021-02-19 by EtchedPixels