EtchedPixels/FUZIX #785: SpectraNet

apocsantos, 2020-04-29:
Hi all,

I've been messing arround with Spectranet (ZX Spectrum ethernet interface), and it would be really cool if fuzix did support spectranet! Any idea if this will ever be possible ?

Regards

EtchedPixels, 2020-04-30:
Sort of a duplicate of #769

closed 2020-04-30 by EtchedPixels

pawosm-arm, 2020-09-26:
@apocsantos what's your experience with Spectranet? It would be great fun to see Fuzix accessed from the network!

apocsantos, 2020-09-26:
A little bit, wrote some code to access files on the network. 

> 
> 
> @apocsantos what's your experience with Spectranet? It would be great fun to see Fuzix accessed from the network!

pawosm-arm, 2020-09-27:
I was wondering if I could find candidates to follow up the discussion on #769. I've managed to control SpectraNet from CP/M Plus on my real +3e (with both 'disable' jumpers closed on SpectraNet) and also managed to do the same on FUSE emulator in +3e mode (it offers workable SpectraNet emulation). Yet the way CP/M Plus for +3 is designed makes it easy and the sheer amount of information you can google about CP/M is enormous making this task even easier. I don't know that much about Fuzix, I've managed to make it boot successfully on my +3e with ZXMMC mass storage and found it really cool, hence I'd love to have ability to telnet into it.