EtchedPixels/FUZIX #786: Booting an NC200 from a Gotek fails with Error: 1A

tjewell99, 2020-05-15:
I'm trying to boot an NC200 from prebuilt floppy image on fuzix.org via Gotek running Flashfloppy (as the floppy drive has failed in this machine). The process starts as I'd expect given the notes in platform-amstradnc/nc200/autoprg.s - 

```
Loading...
LOAD4000.80
LOAD4000.81
LOAD4000.82
Starting bank 80
```

It then goes back to a loading screen again, and does:

```
Loading...
Error: 1A
Press any key
```

I'm not sure what to do next. Any pointers? Many thanks in advance.

EtchedPixels, 2020-05-16:
Not sure. That means it got as far as loading and running something then somehow ended up back in the loader which died (unsurprisingly)

@davidgiven

davidgiven, 2020-05-16:
It's successfully read everything off the disk, so I'm surprised you're not at least getting a startup banner. How big is the SRAM card you're using?

tjewell99, 2020-05-18:
Right now, I don't have an SRAM card, I'm purely trying a floppy boot (via the Gotek).

davidgiven, 2020-05-18:
Ah, that'll be the problem --- Fuzix on the NC200 runs entirely out of the SRAM card's RAM, as the machine only has 128kB of main memory which isn't enough. So what it's done is to load Fuzix into some unpopulated address space. The loader should really check for this but actually detecting the presence of an SRAM card is surprisingly complicated.

If you want something you can run without an SRAM card, I have a CP/M port: http://cowlark.com/cpmish/ It'll let you use a Compact Flash card as a hard drive, too.

tjewell99, 2020-05-18:
That makes complete sense. I've got a SRAM card coming in the post - eventually - so fingers crossed it all works then. I'll have play with CPMish too. Many thanks!

EtchedPixels, 2020-05-18:
@davidgiven it was more that I wanted to play nice - you can in theory run Fuzix on an NC200 SRAM card and not upset the existing OS - in fact you should be able to suspend, move the SRAM card to another NC200 and resume even 8)

davidgiven, 2020-05-19:
Now I've figured out how to use the PCMCIA slot with a Compact Flash card, I wonder whether an internal-RAM-only configuration would be useful for Fuzix --- 128kB is pretty tight. Does Fuzix support processes smaller than 64kB? If not, you'd have to run in single-tasking mode and swap to the CF card.

tinkercnc, 2020-05-19:
> Now I've figured out how to use the PCMCIA slot with a Compact Flash card, ...

Interesting! 
Could you please share more infos about it? 
Would it work with the NC100?

EtchedPixels, 2020-05-19:
You can run with smaller processes. TRS80 model 4 runs with 2 x 32K processes in memory rather than 1 x 64K. For 4x16K you can use chmem to set process sizes as well.

That said a CF card runs about the same speed as main memory, and Fuzix is carefully optimized to run well with one process in memory at a time (after all that is how the original UZI worked). It's not ideal but it does work, especially with CF, and most spinning rust (because the typical drive has several megabytes of cache so running off one you are basically running off a ramdisc with hardware backup)

davidgiven, 2020-05-19:
@tinkercnc I don't have an NC100, but it seems to use the same interface as the NC200, so probably? But it wouldn't be very useful. The stock ROM doesn't know how to access a CF card and you can't boot from it. On the NC200 you can boot from the floppy and that will then talk to the CF card, but the NC100 can't do that. See http://cowlark.com/cpmish/arch/nc200/README.html#using-a-hard-drive for details.

@EtchedPixels That might be worth pursuing then, as an alternative build option; SRAM cards are small and increasingly hard to come by, and even harder to write to unless you have an old laptop. Being able to use an ordinary CF card seems valuable even if it reduces the available memory.

EtchedPixels, 2020-05-19:
I'm also not sure the NC100 can access the config space and stuff.

tjewell99, 2020-06-14:
A quick update, as I've eventually got hold of a working SRAM card. It now boots very nicely to a bootdev: prompt, where I enter '256'. As I'm using a Gotek, I now swap to the 720K Z80 image I downloaded from fuzix.org and enter '256'. It says it has mounted the image OK, but then panics with "no /init".

I'm thinking I should take the Gotek out and replace the floppy and do this the traditional way to reduce the variables. But if anyone has any thoughts?

tjewell99, 2020-12-28:
Another update. I've replaced the original floppy, and now have two floppies. One is the boot disk, which still boots nicely to a bootdev: prompt. I then switch to the 720K Z80 image floppy, and it still panics. So it looks like this isn't the Gotek causing the problem after all. Is it possible there's something up with the image that's up for download?

EtchedPixels, 2021-06-22:
The NC100/200 had bitrotted somewhat. I've checked in a few patches and todays build now works for me with floppy boot/pcmcia root correctly. The NC100 is also back working except for the NMI suspend/resume bit which I need to investigate further.

closed 2021-06-22 by EtchedPixels