EtchedPixels/FUZIX #787: 128k +2 and DIVMMC EnJOY lock up on boot

Spoozilla, 2020-05-17:
I'm having difficulty getting to a sh prompt on my grey 128k +2 with a DIVMMC-EnJOY v2014. The initial signs look promising.

```
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2019 Alan Cox <alan@etchedpixels.co.uk>
Devboot
128kB total RAM, 64kB available to processes (15 processes max)
Enabling interupts ... ok.
IDE drive 0: ide error, status=00FF
ide error, code=00FF
IDE drive 1: ide error. status=00FF
ide error, code=00FF
SD drive 0: hda: hda1 hda2 hda3 (swap)
SD drive 1: no card found
bootdev: hda2
Mounting root fs (root_dev=2, ro): OK
Starting /init
10 buffers added
init version 0.9.0ac#1
_
```
However, after this the border flashes briefly, at which point the activity light on the DIVMMC remains solidly on and the machine completely hangs. Reseting via the reset switch on the speccy or the DIVMMC resulting in a white border with a black screen. A power cycle is required to bring the machine back to life,

I've managed to build 0.3.0 from source (git checkout cacbe52)  with the patched SDCC as mentioned in a [previous post](https://github.com/EtchedPixels/FUZIX/issues/756) and generated the 32m root file system image,

The sd card is partitioned with a 128M FAT32 partition with ESXDOS 0.8.7 installed. I then have the 32MB type 7e partition and a 4MB typr 7f partition for swap. I have tried 4 different sd cards just in case. I've also tried dd'ing the z80 32m rootfs image from fuzix.org but all to no avail.

Could this be a compatibility issue with this particular DIVMMC interface?

EtchedPixels, 2020-05-19:
Sounds like #766

There is something about the way the Sinclair DIVMMC and related hardware is designed that disagrees with the Fuzix SD driver and causes a lock up on disk write. The MMC I have is built into my ZX-UNO and that one works as does the emulation. It seems that whatever magic is hidden in the ZXMMC and DIVMMC-Enjoy is a bit different.

Spoozilla, 2020-05-20:
As you point out there are definitely some similarities with [#766](https://github.com/EtchedPixels/FUZIX/issues/766). Although not entirely comparable hardware, zxmmc and +3e roms are very different to DIVMMC, there are some interesting changes in the diffs there. i'll have a crack at applying them and report back.

Spoozilla, 2020-05-20:
I applied the changes/diffs presented in [#766](https://github.com/EtchedPixels/FUZIX/issues/766) with the noop SD timing(?) changes to divmmc.c instead of zxmmc.c. The system boots and I've run some fscks and dd's which are fine.  I'm happy to close this issue and track 766 from now on.
```
diff --git a/Kernel/dev/devsd.c b/Kernel/dev/devsd.c
index f656ac7a..0f9bcdc8 100644
--- a/Kernel/dev/devsd.c
+++ b/Kernel/dev/devsd.c
@@ -48,6 +48,8 @@ uint_fast8_t devsd_transfer_sector(void)
                     sd_spi_transmit_byte(0xFF); /* dummy CRC */
                     sd_spi_transmit_byte(0xFF);
                     success = ((sd_spi_wait(false) & 0x1F) == 0x05);
+                   if (success)
+                       while (sd_spi_wait(false) == 0x00) { }
                 }
             }
        }else
@@ -73,12 +75,10 @@ void sd_spi_release(void)
 
 uint8_t sd_spi_wait(bool want_ff)
 {
-    unsigned int timer;
+    int repeats = 12800;
     uint_fast8_t b;
 
-    timer = set_timer_ms(500);
-
-    while(true){
+    while(--repeats){
         b = sd_spi_receive_byte();
         if(want_ff){
             if(b == 0xFF)
@@ -87,10 +87,6 @@ uint8_t sd_spi_wait(bool want_ff)
             if(b != 0xFF)
                 break;
         }
-        if(timer_expired(timer)){
-            kputs("sd: timeout\n");
-            break;
-        }
     }
 
     return b;
```
```
diff --git a/Kernel/dev/zx/divmmc.c b/Kernel/dev/zx/divmmc.c
index 537e143f..19ff47ad 100644
--- a/Kernel/dev/zx/divmmc.c
+++ b/Kernel/dev/zx/divmmc.c
@@ -66,6 +66,10 @@ doread:
     inir
     ld a,(_vtborder)
     out (0xfe),a
+    in a, (0x3f)
+    nop
+    nop
+    in a, (0x3f)
     pop af
     or a
     jp nz,map_kernel
```

closed 2020-05-20 by Spoozilla