EtchedPixels/FUZIX #792: Z80 kernel fails to compile with SDCC version 4.0

jbquirk, 2020-06-09:
Not sure if this has been reported but I built SDCC Version 4.0 from the sources and FUZIX build fails with this error:
sdcc -c --std-sdcc99 --no-std-crt0 -mz80 -I/home/john/Desktop/fuzix/FUZIX-master/Kernel/cpu-z80 -I/home/john/Desktop/fuzix/FUZIX-master/Kernel/platform-rc2014 -I/home/john/Desktop/fuzix/FUZIX-master/Kernel/include --max-allocs-per-node 30000 --opt-code-size --stack-auto --constseg CONST --peep-file /home/john/Desktop/fuzix/FUZIX-master/Kernel/cpu-z80/switch.peep  --external-banker -I../dev/ -I../dev/net/ -I../dev/rc2014/ --codeseg CODE1 -c main.c
at 1: warning 117: unknown compiler option '--external-banker' ignored
_**main.c:159: error 98: conflict with previous declaration of 'strlen' for attribute 'type' at /home/john/Desktop/fuzix/FUZIX-master/Kernel/cpu-z80/cpu.h:37**_
from type 'int function ( const-unsigned-char generic* fixed) fixed'
  to type 'int function ( const-unsigned-char generic* fixed) fixed'
make[2]: *** [Makefile:40: main.rel] Error 1
make[2]: Leaving directory '/home/john/Desktop/fuzix/FUZIX-master/Kernel/platform-rc2014'
make[1]: *** [Makefile:106: target] Error 2
make[1]: Leaving directory '/home/john/Desktop/fuzix/FUZIX-master/Kernel'
make: *** [Makefile:89: kernel] Error 2  

I down loaded and built version 3.8 and complies fine I just need to fix the external banker issue. 
which is why I was building SDCC from source. 
works:
```
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.8.0 #10562 (Linux)
published under GNU General Public License (GPL)
fails 
```
SDCC : ```
mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ez80_z80/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8/pdk13/pdk14/pdk15 4.0.0 #11528 (Linux)
published under GNU General Public License (GPL)

```
Further info I built the whole package with version 3.8 that came with Ubuntu 20.04 and for this last part did a **make kclean**  and then **make kernel**. Not if that is part of the issue. 

Thanks for the fun project to work on.

EtchedPixels, 2020-06-09:
at 1: warning 117: unknown compiler option '--external-banker' ignored

You need the patched compiler to build the RC2014 kernel as it's banked and I've never been able to get the sdcc people interesting in the banking patches.

I've not tried 4.0, although the error report saying that there is a conflict between two identical definitions does not inspire confidence! I don't move often because moving SDCC means re-porting the patches for banking, and also dealing with the fact that SDCC tends to be somewhat buggy and the bugs change each release. I prefer to have one unknown to fight at a time!

jbquirk, 2020-06-09:
Thanks for the reply just found your version of SDCC so will use that now. I knew about the --external banked option. But your right it does seam very odd that an identical definition is flagged as an error. 
Still finding my way around fuzix and having lots of fun hopeful can contribute something useful soon.

spth, 2020-08-10:
> I've not tried 4.0, although the error report saying that there is a conflict between two identical definitions does not inspire confidence! I don't move often because moving SDCC means re-porting [...]

Where can I find the exact declarations, so I can try to reproduce the SDCC issue without having to compile all of FUZIX?

EtchedPixels, 2020-08-10:
rebasing to 4.0 is on my list so I'll investigate that and some others then - however I'm currently taking a break from all the coding stuff

jbquirk, 2021-03-29:
I also had issues with the the version of sdcc for fuzix so I strongly suspect my build environment for both compilers had issues. I haven't had a chance to revisit this as I've been distracted by another project. I plan to revisit this soon with a fresh rebuild of the compiler and see what happens.

EtchedPixels, 2021-03-29:
Every new sdcc tends to introduce new weird corner case bugs - and sometimes finds Fuzix 'just by luck' things too. The sdcc bugs usually  get fixed promptly it does mean that it's best to stick to the exact tested versions for Fuzix builds.

closed 2021-07-05 by EtchedPixels