EtchedPixels/FUZIX #793: Fuzix 0.4pre issues on emulated RC2014

jbquirk, 2020-06-14:
Well I finally got all the tools in place and finally got a successful build.
The issue I have now is it runs so slow on the rc2014 emulator.
At first I thought it had crashed but it was just so slow in coming up.
I edit init and dropped myself in to a shell had a look around file system OK .
So disk creation process seemed to be OK. I changed the config file to try and build a version that dropped of the offending package but nothing seemed to work. Shot gunning I know 
I am definitely running version 4.0 output from:
**# uname -a
Fuzix 0.4pre1  rc2014**  

Another confirmation that I am running .4 is I put all the version .4 binaries on the disk when I had version 3 running and the binaries didn’t work as I expected once I had .4 up they run fine.

Things like this run just OK **“man sh”** it is slow but it gets there in the end **“man invalid”** seems to be stuck and I interrupt it. 
Something is obvious getting stuck in a time sucking loop but the system seems to be robust enough to not die so this is good but I’m now not sure where to look next.

This is how I built my disks.

**dd if=0.3_m.ide of=foo.raw skip=2**

This disk is a copy of the version .3 disk found on the website.

**sudo losetup -P /dev/loop30 foo.raw
ls -l /dev/loop30***

Last step just to confirm that disk seems valid and all partitions a found

 **cd Standalone/filesystem-src/
 ./build-filesystem fuzix.dsk 256 65535**
  
Use tool to create a root file-system on a raw disk image

**sudo dd if=fuzix.dsk of=/dev/loop30p1**

Put the image onto cf card image.

 **sudo ./ucp /dev/loop30p1 
 sudo ./fsck /dev/loop30p1** 
 
A couple of quick checks to confirm all is OK

next from the kernel/platform-RC014 directory

**dd if=fuzix.romwbw of=/dev/loop30 seek=2 conv=notrunc
sudo losetup -d /dev/loop30**

Stop loop device 
Move to directory where raw disk image is located.

Then rebuild RC2014 emulator disk 

**dd if=foo.raw of=0.3_m.ide  seek=2 conv=notrunc**

now run in emulator
 
**./rc2014 -f -b -r RC_2.512k.rom  -i 0.3_m.ide  -s  -w -**R -c

Note I do use -f otherwise it would never boot in a reasonable time.

```
C2014 Z80 @ 7.372MHz
0 MEM W/S, 1 I/O W/S, INT MODE 1
512KB ROM, 512KB RAM

SIO0: IO=0x80 SIO MODE=115200,8,N,1
SIO1: IO=0x82 SIO MODE=115200,8,N,1
MD: UNITS=2 ROMDISK=384KB RAMDISK=384KB
IDE: IO=0x10 DEVICES=1
IDE0: 8BIT LBA BLOCKS=0x00040000 SIZE=128MB
PPIDE: IO=0x20 NOT PRESENT

Unit        Device      Type              Capacity/Mode
----------  ----------  ----------------  --------------------
Disk 0      MD1:        RAM Disk          384KB,LBA
Disk 1      MD0:        ROM Disk          384KB,LBA
Disk 2      IDE0:       Hard Disk         128MB,LBA
Serial 0    SIO0:       RS-232            115200,8,N,1
Serial 1    SIO1:       RS-232            115200,8,N,1

RC2014 Z80 Boot Loader

Boot: (C)PM, (Z)System, (M)onitor,
      (L)ist disks, or Disk Unit # ===> 
Reading disk information...

LOC=F200 END=F400 ENT=F200 LABEL=Fuzix WBW Loader
Loading...FDC DOR 0C [AT/EISA DMA DSEN DSEL0]
FUZIX version 0.4pre1
Copyright (c) 1988-2002 by H.F.Bower, D.Braun, S.Nitschke, H.Peraza
Copyright (c) 1997-2001 by Arcady Schekochikhin, Adriano C. R. da Cunha
Copyright (c) 2013-2015 Will Sowerbutts <will@sowerbutts.com>
Copyright (c) 2014-2020 Alan Cox <alan@etchedpixels.co.uk>
Devboot
Z80 CTC detected at 0x88.
DS1302 detected at 0xC0.
tty1: Z80 SIO at 0x0080.
tty2: Z80 SIO at 0x0082.
512kB total RAM, 432kB available to processes (20 processes max)
Enabling interrupts ... ok.
IDE drive 0: hda: hda1 hda2 hda3 (swap) hda4 < > 
IDE drive 1: ide error, status=0000
SD drive 0: no card found
bootdev: hda1

Mounting root fs (root_dev=1, ro): OK
Starting /init
4 buffers added
init version 0.9.0ac#1
init: Opening inittab
init: Opened inittab
```

After typing **hda1** it get here very quickly then slows considerably   
```

Checking root file system.
 ^ ^
 n n   Fuzix 0.3.1
 >@<
       Welcome to Fuzix
 m m

login: 
```

But it does get there.

I hope this helps.

EtchedPixels, 2020-06-17:
Just done a rebuild from clean and without -f I see about a 5 second delay between Checking root file system and login: , which seems about right. I don't see your problem. OTOH I also just built it with the atoi/atol fixes for a stupid C library bug I added so that might be related.

jbquirk, 2020-06-18:
Still learning how it hangs together so don't rule out a build error on my part.  I will do a rebuild tonight with the new library and see if that fixes the issue.  I had init complain about a disk badly un-mounted and  was going to do a check I went off and did something else and  40 minutes it never finished the boot. 
Also as can see I've started to write a detailed disk building guide happy to contribute the final version to the doc files.

EtchedPixels, 2020-06-18:
That would be fsck running a full disk check. That will take some time
but certainly should not have been 40 minutes and you should see a progress indicator too.

(use shutdown to shut the machine down properly)

Testing that scenario I see

bootdev: 
bootdev: 1
Mounting root fs (root_dev=1, ro): warning: mounting dirty file system, forcing r/o.
OK
Starting /init
4 buffers added
init version 0.9.0ac#1
Checking root file system.
Filesystem was not cleanly unmounted.
Device 0 has fsize = 65535 and isize = 256. Continue? y
Pass 1: Checking inodes...
Pass 2: Rebuilding free list...
Rebuild free list? y
s_tfree was changed to 58004 from 58005.
Pass 3: Checking block allocation...
Pass 4: Checking directory entries...
Pass 5: Checking link counts...
**** Root filesystem was modified, immediate reboot required.

and it takes about 60 seconds without -f

jbquirk, 2020-06-18:
I did a rebuild and still have same issue - so that leaves something in my setup or something in how I built the emulator. I will have get a real Z80 system and try it out. But for now i will investigate and see if can find what the issue is.

jbquirk, 2020-06-18:
I now feel the issue is in the RC2014 simulator as in normal it consumes 20% CPU and in fast 100%.
The other odd thing and it points to a bad build of the emulator is the date command under fuzix reports. Crazy fast time see these
date;date
Wed Jun 27 16:50:49 GMT 3381
Wed Jun 27 16:51:42 GMT 3381
What has thrown me it is Fuzix that seems to trigger the bug. 

Ah the joy of building from source..

jbquirk, 2020-06-18:
I have investigated more and tried switching off CTC and RT these made no real difference. I feel the clue is in the crazy fast time. This fast time also happens at normal simulated speed not supper fast. 
I switched off the CTC and RT via the command line switches.

EtchedPixels, 2020-06-18:
In fast mode the emulator runs flat out so 100% is normal.

In slow mode it does some CPU cycles and takes a nap. Is the date clock crazy when you are running without -f, if so that would imply something is causing an interrupt storm.

jbquirk, 2020-06-19:
Yes when I ran the emulator in slow/normal mode and the the clock results are similar i.e. crazy fast. I haven,t tried the ACIA or SIO options yet.  Will spend some more time on this tonight I want to get this to work as I loved the old Z80 back when I started out in computing. I agree it has to be an interrupt storm of some sorts triggered by something fuzix does as the ROMWBW stuff all seems to work and respond as expected. Its all fine until it loads init off the disk.  From the messages I put into init its fine even starting to read the init files. So i'm thinking of putting some more debugging messages to see what triggers it or seeing if I can insert some sort of trap code in the emulator to look around at the state of the emulated Z80. I am currently running the Ubuntu 20.04 with the stock GCC that comes with that distribution.

jbquirk, 2020-06-21:
I decided to rebuild completely from from a clean clone of the repository and deleted the contents of /opt/fcc in case during my convoluted attempts to build fuzix I had some incompatible binary files left lying around. So the only change I made was I edited the the top level makefile to set the platform to RC2014 and then saved. I then did make, on my system a few hours later it finished, sadly no different to before after rebuilt the disk will all the newly built software. So I'm still not sure where to start looking for the error so I've attached my fuzix binary so you compare it to yours to see if mine is different. The only change I have made to the repository was the previously mentioned make file change I copied the repository on Thu 18 Jun 2020 18∶55∶03 (Local time UTC+10). Also vi inside fuzix also crashes and writes rubbish to the disk.
[fuzix.romwbw.zip](https://github.com/EtchedPixels/FUZIX/files/4808887/fuzix.romwbw.zip)
 My init experiments showed it was fine up till it went of to do the init_boot section.

jbquirk, 2020-06-27:
Sorry Didn't answer the question on the SIO and ACIA they made no difference - I have had to take a break for chasing this I will hopefully get a chance in about two weeks.

EtchedPixels, 2020-06-30:
Did a quick test - with your kernel image I see it misbehaving. Swap it with my kernel build and it runs.

Can you attach the matching fuzix.map when you get a chance then I can try and see what it is doing all that time

jbquirk, 2020-07-07:
Thank you for that I now know it must something in build environment :( Here is the Map file and hogs files as well
[fuzix.map.zip](https://github.com/EtchedPixels/FUZIX/files/4884244/fuzix.map.zip)

EtchedPixels, 2020-07-08:
Thanks. I will have a dig when I get some time

EtchedPixels, 2021-06-22:
Never managed to figure this out, or reproduce it with my builds. Now old so closing.

closed 2021-06-22 by EtchedPixels