EtchedPixels/FUZIX #795: sc130 init panic.

Spoozilla, 2020-06-27:
I can't get my sc130 to boot any version of Fuzix aside from the sc126 downloadable image on fuzix.org. The current tree (96fa4f4) compiles perfectly and the loader seems function correctly, however once supplying the correct bootdev the system crashes and requires a reset.
```
Mounting root fs (root dev=1, ro): OK
Starting /init

panic: killed init
monitor: SP=F7EE, PC=64BD
:( 
```
I've tried using the 0.4pre1 filesystem from fuzix.org and also building one from the tools in /Standalone/filesystem-src. I have also tried using a compact flash adapter instead of the SPI/SD. Both produce the same error.

EtchedPixels, 2020-06-27:
I don't have an SC130 so I have no idea what is going on. It's currently not a supported platform. There shouldn't be anything in Fuzix that blows up because of the small hardware differences from the SC126 so I have no idea what is going on and no real opportunity to investigate. Could be a ROMWBW funny even. It's a known incompatibility though and you are not the only reporter.

Spoozilla, 2020-06-27:
The romwbw configuration between the two devices seems to boil down to the LED status support and one extra SDCNT setting that limits the sc126 to a single SD device. Seeing as this should be the same for the sc130 I'll start investigating that config. Do you happen to know the commit that relates to the working build you released for the sc126 image? Comparing the init between the two commits might give me something else to go on.

mennozon, 2020-06-28:
I can confirm the panic on sc126 as well so I don't think the difference between sc130 should be the reason.

I've enabled syscall debug and it panicked after syscall 30 but I'm not sure if this is helpful, I can provide the full output tonight if needed.

EtchedPixels, 2020-06-28:
Which ROMWBW do you have on the SC126 ? I'm wondering if that's actually the difference and some of the Fuzix boot abuse of ROMWBW is breaking the new ROMWBW. I'll try and get back to this a bit next week and boot it again on my SC126

mennozon, 2020-06-28:
My SC126 kit came with RomWBW HBIOS v3.0, 2020-03-27.

If it helps, I've tried a few more times tonight and sometimes it panics with "corrupt inode". As @Spoozilla mentioned, I've also tried with both SD card and CF card using the RC2014 CF card module and the panics look somewhat similar.

I say somewhat similar because with syscall debug enabled it seems a bit random how far it gets before panicking.

Spoozilla, 2020-06-28:
Just to complete the picture I've used HBIOS v3.0 2020-03-27 but I also flashed a custom configuration of 3.0.1 2020-03-30 to enable my ds1302 RTC card.

EtchedPixels, 2020-06-30:
Ok duplicated - including with older ROMWBW. Will take a look properly when I get some time and see what I broke

EtchedPixels, 2020-06-30:
Try that

Spoozilla, 2020-07-01:
This has improved the situation. init runs correctly and I can log in. However any interaction with the file system results in an sd: timeout notification. This doesn't happen if I use the compact flash. I have another SD card arriving this afternoon which should rule out a problem with the media.

EtchedPixels, 2020-07-01:
vi is unrelated.  I think I know what the vi problem is but please file  bug for it so I remember. It should be telling you out of memory not crashing but I think there is a sign handling error somewhere in the file loading.

EtchedPixels, 2020-07-01:
other thing to check is whether commenting out the timeout check in sd_spi_wait() fixes it.

Spoozilla, 2020-07-01:
I changed sd_spi_wait() in Kernel/dev/devsd.c to match the [changes](https://github.com/EtchedPixels/FUZIX/issues/787#issuecomment-631602653) I had to implement for the zx spectrum +2/divmmc to work and this seems to have removed the warnings.

EtchedPixels, 2020-07-01:
There is a similar bug report on the ZXMMC interface and I can also it seems fake an error by running my finger over the pins during an I/O at which point I see sd: failed, retrying and then it hangs whereas it ought to try 8 times.

EtchedPixels, 2020-07-01:
Yes I was looking at the timer code and resulting asm. I don't see anything wrong in my build so it's weird that either the card is taking so long or the timeout is happening too fast. If you put the timer check back partially how many iterations do you log before a timeout ?

Spoozilla, 2020-07-02:
I have to admit I'm a little out of my depth here. I'm not much of a programmer and my C skills are frankly shocking. 

I put some rather ham-fisted code into the sd_spi_wait() procedure to return the amount of remaining iterations of the (--repeats) loop just before the procedure returns b. Obviously this produced a lot of results, so I wont post them here but following a successful login and ls -la, of the many hundreds of results all of them were completed within 35 iterations with 99% being within 5. 

I reverted back to the original timer code, minus the timeout break; and added my code back in to dump (timer) to the console on the return of b. This seems to just return an incrementing value. However, interestingly with this "debugging" enabled there were no timeout errors.

I then restored the original code and increased the timer value from 500 to 5000 and it seems to have had no effect on the frequency of the timeouts.

I'm not sure if any of this was useful but I can provide the small changes I made and the logging if required.

EtchedPixels, 2020-07-04:
This should give you just data when it would have timed out. I'm still not sure what is going on here, it's very odd indeed.
[1.txt](https://github.com/EtchedPixels/FUZIX/files/4872995/1.txt)

EtchedPixels, 2020-07-04:
Try that again - this patch should hopefully log the timeout cases only and how many iterations were involved - something odd is going on here though and I can't make it happen on my box.
[1.txt](https://github.com/EtchedPixels/FUZIX/files/4872997/1.txt)

Spoozilla, 2020-07-06:
Ok, I have applied the patch and tested with the 2 SD cards I have to hand. The first is a 32GB SanDisk Ultra HC I and the other is a 16GB Verbatim Class 10 UHS-1 U1. I ran through the boot process 5 times on each and have attached some logs. Once the sd:timeout error triggers, the system needs reseting.
[verbatim_sd.txt](https://github.com/EtchedPixels/FUZIX/files/4878424/verbatim_sd.txt)
[sandisk_sd.txt](https://github.com/EtchedPixels/FUZIX/files/4878425/sandisk_sd.txt)

EtchedPixels, 2020-07-06:
Very strange - so once the timeout triggers on your system the card never recovers, it doesn't even exit the retry loop, yet when you change it to just retrying it works.

Spoozilla, 2020-07-06:
Indeed, that was pretty much my understanding of what was occurring. The fact that it doesn't seem to have any issues with the compact flash (which I have just tested again) would suggest to my naive brain some sort of adverse relationship between the timer system and the SPI/SD bit banging?

EtchedPixels, 2020-07-06:
That was partly what I was assuming except that the SC126 currently doesn't use the RTC (which does share bits with the SD card chip selects) and the similar behaviour you reported in the ZX case there are no shared things.

The actually timer_ routines called should do nothing more than set a 16 bit value to tenths of a second beyond now, and compare it with the current number, at least on the SC126.

Spoozilla, 2020-07-06:
Yes, very odd. It does feel like I am now the only constant in the equation. Maybe it's something I'm doing in build process? I could rebuild my entire tool chain on another machine/OS and see if that still produces the issue.

mennozon, 2020-07-06:
I'm seeing the same as @Spoozilla on my sc126, once it hangs it doesn't seem to recover.

I did try to eject the sdcard while this happened and that seems to exit the loop but once the card is back it will hang again on the next timeout, I've attached a boot log in case that might be of use.

[boot.txt](https://github.com/EtchedPixels/FUZIX/files/4879853/boot.txt)

Spoozilla, 2020-07-08:
I started from scratch using FreeBSD 12.1 and port compiled gcc 9.3/sdcc 3.9 with your patches. I used a completely different SD card (8gb Transcend). I created the image with the follow method.

```
gmake kernel
cd Kernel/platform-sc126
dd if=/dev/zero of=fuzix.img bs=512 count=2
cat fuzix.romwbw >> fuzix.img
dd if=fuzix.img of=/dev/da0
fdisk-linux /dev/da0
n, p, 1, 2048, +32m, t ,7e
n, p ,2, 67584, +4m, t, 2, 7f, w
cd ../../Standalone/filesystem-src
./build-filesystem rootfs 256 65535
dd if=rootfs of=/dev/da0p1
```

The results are functionally identical with the previous builds. Since you're not seeing the error and @mennozon is the only other person reporting the issue could this be a hardware error? Something defective in the SPI to SD interfaces we have?

EtchedPixels, 2020-07-08:
When I get a bit of time I'll upload an SD card image of the one that works on my system.

I don't think it's hardware, and if your SD card works reliably in CP/M on that machine then even if it is hardware there is clearly something software that can be done to avoid the problem - which in reality makes it a software problem.

There are differences in the SD adapters (some have pull-ups some don't) but the code is supposed to deal with both and afaik mine is the one without - which should be the worst case. I have some more adapters on order so I will try switching them around.

oholiab, 2020-07-26:
FWIW I've just managed to get FUZIX working on my SC130 with RomWBW v3.0. built against current master (af6bd7bd19dcd12206da7bf5070d979c6de8530e) with very minimal changes (just TTY baud rate) and I'm also getting output littered with `sd: timeout` messages although it doesn't seem to be causing any problems in the actual functioning of the system. I can edit and read files perfectly fine.

I'm more than happy to test out changes on this platform if it helps :)

closed 2021-07-09 by EtchedPixels