EtchedPixels/FUZIX #796: vi write issue

Spoozilla, 2020-07-01:
vi appears to be writing files full of garbage.
```
# vi .profile
i
#!/bin/sh
TERM=xterm ; export TERM
[esc]
:w
:q
# ls -la
drwx--x---	2	root	root		512	Jul 01 15:16 .
drwxrwxrwx 11	root	root		512	Jul 01 15:10 ..
-rw-r--r--  1	root	root	     64290 Jul 01 15:16 .profile
# head .profile
�!                                                         �!�9�n&3�!9~�      �(
!��͊*�!�!��!�����!���!�readshort read from tchelp.!��
������:*�MD!9~#fo       #��I*!��9�!9�u��t���b$��|(!��͊*�!�����l$MDy< x< !��     !
#
```
Creating the same file with echo.
```
# echo "#!/bin/sh" > .profile
# echo "TERM=xterm ; export TERM" >> .profile
# ls -la
drwx--x---	2	root	root		512	Jul 01 15:33 .
drwxrwxrwx 11	root	root		512	Jul 01 15:10 ..
-rw-r--r--  1	root	root	     35 Jul 01 15:33 .profile
# head profile
#!/bin/sh
TERM=xterm ; export TERM
```
Editing an existing file in bi

```
# vi .profile
j
j
i
echo "now it's broken"
[esc]
:w
:q
# ls -la
drwx--x---	2	root	root		512	Jul 01 15:34 .
drwxrwxrwx 11	root	root		512	Jul 01 15:10 ..
-rw-r--r--  1	root	root	     60473 Jul 01 15:34 .profile
# head .profile
�!                                                         �!�9�n&3�!9~�      �(
!��͊*�!�!��!�����!���!�readshort read from tchelp.!��
������:*�MD!9~#fo       #��I*!��9�!9�u��t���b$��|(!��͊*�!�����l$MDy< x< !��     !
# 
```

EtchedPixels, 2020-07-02:
Fixed.. sign overflow

closed 2020-07-02 by EtchedPixels

Spoozilla, 2020-07-02:
Sorry but this is not fixed. I just checked out the repository and did a clean build. I'm afraid it's still behaving as above.

Spoozilla, 2021-01-04:
I've just been playing about with this again. The last commit where vi works as expected when saving was commit [b99f8e0](https://github.com/EtchedPixels/FUZIX/commit/b99f8e0605f4676fa9eca269f24cd50b0b6db37c).